realsenseai/librealsense #6882: Resolve stream profile from the sensonr

afraidofdark, 2020-07-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (05.12.03.00) | 
| Operating System & Version |   {Win 10| 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.29 }                          | 
| Language                            |  {c++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I am working with D400 series depth cameras and using  rs2 lib.
I want to use syncer to software sync multiple camera streams provided from different devices. (Different cameras)
In this [thread](https://github.com/IntelRealSense/librealsense/issues/2656), there is a discussen of how to use syncer. But on the code, first stream profile on the sensor has been used. I wan to be able to set it as I do here

```
rs2::config* cfg;
cfg->enable_stream(RS2_STREAM_DEPTH, -1, 1280, 720, RS2_FORMAT_Z16, 30);
pipe->start(*m_Cfg);
```

but with a syncer.

```
rs2::sensor sensor;
rs2::stream_profile colorProfile = sensor.get_stream_profiles()[0]; // Instead I wan to query a stream expilicitly. Like enable_stream(RS2_STREAM_DEPTH, -1, 1280, 720, RS2_FORMAT_Z16, 30);
sensor.open(colorProfile);
sensor.start(m_Sync);
```

It would be very nice if you show an example code which configure every device and sensor as it done with pipe examples and use it with a syncer and provide a method to measure syn error between coherent frames if there is any.

dorodnic, 2020-07-21:
Hi @afraidofdark 
All logic related to choosing stream profiles is localised in [resolver.h](https://github.com/IntelRealSense/librealsense/blob/master/src/pipeline/resolver.h)
It is part of implementation, not public API, so it won't work outside the SDK, but it can give you a hint about what pipeline is doing internally.

afraidofdark, 2020-07-22:
I solve it by iterating over all of the stream profile. There should be a look up function in the library.

```
std::vector<rs2::stream_profile> stream_profiles = sensor.get_stream_profiles();
for (rs2::stream_profile& sp : stream_profiles)
{
// Check settings.
rs2::video_stream_profile vsProf = sp.as<rs2::video_stream_profile>();
if (vsProf.width() != 1280)
{
    continue;
}
if (vsProf.height() != 720)
{
    continue;
}

if (vsProf.fps() != 30)
{
    continue;
}
                                                                                
if (vsProf.format() != RS2_FORMAT_Z16)
{
    continue;
}
}
```

RealSenseSupport, 2020-07-27:
@afraidofdark Glad to hear the issue resolved. Any other questions about this? Thanks!

closed 2020-07-27 by afraidofdark