realsenseai/librealsense #6883: Unable to install pyrealsense2 in docker (windows)

Harleens13, 2020-07-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   | Intel RealSense Viewer v2.35.2 | 
| Operating System & Version | Windows 10   | 
| Kernel Version (Linux Only)    |       | 
| Platform                                 | PC  |
| SDK Version                           | 2.0 | 
| Language                              |  python 3.7 | 
| Segment			        | { }  | 

### Issue Description

I wish to install pyrealsense2 in dockerfile. `pip3 install pyrealsense2` gives the follwoing error -

[image]

I am a beginner in docker and docker file is so written - 
[image]

Can somebody kindly point me in the right direction and say what is wrong?
I was able to run and use pyrealsense2 in my local computer and installed it using  - `pip install pyrealsense2`.

I've been reading a lot to make from [source](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source) but I've been unsuccessful due to my incompetency and other results are for Linus whereas my host OS is windows 10.
Help appreciated.
Thank you in advance.

dorodnic, 2020-07-21:
Hi @Harleens13 
Could you please check version of python? 
We currently publish packages for 2.7, 3.6 and 3.7, but not for 3.8.

Harleens13, 2020-07-21:
Hello @dorodnic 
As I see.. its 3.7.3

[image]

But what you think it was 3.8 again? Also in case someone has more than one python version installed how could we make sure it downloads to version 3.7?

ev-mp, 2020-07-21:
@Harleens13 , the downloaded packages are targeted for python 3.8, so apparently the docker has this version installed
[image]

Harleens13, 2020-07-22:
Hello @ev-mp ,
Thank you for pointing out. That was very foolish of me to not understand that part.
We will degrade the version but either way, is builiding from source still an option, yeah?

closed 2020-07-22 by Harleens13