realsenseai/librealsense #6885: repository at aws has missed packages fro arm64

AndreV84, 2020-07-21:
where to get arm deb for calibration?
 am trying to do calibration steps under linux using the official instruction steps from
https://dev.intelrealsense.com/docs/intel-realsensetm-d400-series-calibration-tools-user-guide
It won't work.

sudo apt-get install librscalibrationtool
Unable to locate package librscalibrationtool

RealSenseSupport, 2020-07-27:
@AndreV84  Did you do the following steps according to https://dev.intelrealsense.com/docs/intel-realsensetm-d400-series-calibration-tools-user-guide#section-3-1-2-1-debian-package-installation-through-amazon-aws ?  Also what's your arm platform?

echo 'deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main' | sudo tee /etc/apt/sources.list.d/realsense-public.list
sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
sudo apt-get update

AndreV84, 2020-07-27:
Thank you for your response!
my arm platform is nvidia Jetson [various models including nano, nx, agx] 
yes, I got through the steps in the tutorial;
on x86_64 there is the package; on arm64 there is no such package, but librealsense2 still presented

AndreV84, 2020-07-27:
```
sudo apt-get update
Hit:1 https://repo.download.nvidia.com/jetson/common r32 InRelease
Hit:2 https://repo.download.nvidia.com/jetson/t194 r32 InRelease               
Hit:3 http://ports.ubuntu.com/ubuntu-ports bionic InRelease                    
Hit:4 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease      
Get:6 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease [3,230 B]
Hit:7 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease          
Hit:8 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease           
Get:9 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main arm64 Packages [5,948 B]
Fetched 9,178 B in 2s (5,740 B/s)                   
Reading package lists... Done
 sudo apt-get install librscalibrationtool
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package librscalibrationtool
```

RealSenseSupport, 2020-07-31:
@AndreV84  Sorry, got confirmation from our engineer team that the DC packages for arm64 not uploaded to AWS yet. They will upload soon. Will update you when upload is finished. Sorry again for the trouble.

RealSenseSupport, 2020-09-03:
@AndreV84 The DC packages for arm64 now uploaded to AWS now. Please have a try. Looking forward to your update. Thanks!

AndreV84, 2020-09-03:
thank you for the update!

closed 2020-09-03 by AndreV84