realsenseai/librealsense #6886: How to convert (u,v) to (x,y,z) using python

EvanZouQ, 2020-07-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC.|                 | 
| Language                            |  python  | 
| Segment			|  Object Detection | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Dear Sir or Madam, I can successfully perform the example"distance_to_object-checkpoint.ipynb". And I want get the (x,y) coordinate of this detected distance. What should I do?  Thanks a lot.

Leo16112, 2020-07-21:
Hello,
you need transform depth to world coordinates.
One of the ways to do it in python is exampled here:
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py

You need lines beginning with line 292
Kind regards

Dilip-Gurumurthy, 2020-07-22:
Checkout  #6749

EvanZouQ, 2020-07-23:
> Hello,
> you need transform depth to world coordinates.
> One of the ways to do it in python is exampled here:
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py
> 
> You need lines beginning with line 292
> Kind regards

Thanks. I will try it.

RealSenseSupport, 2020-10-05:
Hi @EvanZouQ Do you need further help on this? If we don't hear from you in 7 days, this issue will be closed.

closed 2020-10-13 by RealSenseCustomerSupport