realsenseai/librealsense #6887: Playback from bag files results in different number of frames inspite of set_real_time=False

vidullan, 2020-07-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } | 
| Operating System & Version |   Ubuntu 18  | 
| Kernel Version (Linux Only)    |  5.3 | 
| Platform                                 | PC |
| SDK Version                            |  2.34.0.1454                          | 
| Language                            |  python3                          | 

### Issue Description

I've recorded bag files using the real sense viewer in Windows 10 and am trying to extract frames from it in python3 in ubuntu. 

The issue is even though I'm in playback mode with real time set to false I get inconsistent number of frames read from the bag file. For example, on the exact same file on different runs I get anywhere between 728 frames and 300 frames.

Is there no way to read all the available frames consistently since this is a pre-recorded file? The information is obviously there as rerunning the same snippet seems to extract all the frames sometimes. 

(I'm using the timestamp to determine end of file since using repeat_playback=false means wait_for_frames will hang)

Here is the relevant code snippet:

# Create pipeline
pipeline = rs.pipeline()
# Create a config object
config = rs.config()
config.enable_device_from_file(bag_path)
profile = pipeline.start(config)
playback = profile.get_device().as_playback()
playback.set_real_time(False)

while True:
    frames = pipeline.wait_for_frames()
    ts = frames.get_timestamp()
    print(ts,end='\r')

    if ts < ts_prev:
        break
    else:
        ts_prev = ts

MartyG-RealSense, 2020-07-22:
Hi @vidullan   You are not the first person to receive inconsistent frame counts from a bag file when reading it with Python.  During my research into your problem, the most common advice for this problem from RealSense support team members was to check the frame count with a program called **rosbag-inspector** to find out what the correct number of frames in the bag file should be when reading the count with your own Python code.

https://github.com/IntelRealSense/librealsense/tree/master/tools/rosbag-inspector

One of the support team members suggested the following method for checking frame count with rosbag-inspector:

"Open the target bag file, and look at the **topics /device_0/sensor_0/depth_0/image/data** or **/device_0/sensor_1/color_0/image/data** and look at the count".

vidullan, 2020-07-23:
Thanks for the reply @MartyG-RealSense and the link.

Upon further investigation, the problem seems to be the API repeating frames an arbitrary number of times. Using the timestamp associated with each frame to reject duplicate frames seems to generate the correct number of frames consistently.

MartyG-RealSense, 2020-07-23:
I'm pleased that you found a solution.  Do you require further assistance, please?

vidullan, 2020-07-23:
@MartyG-RealSense If possible could you shed insight into why I might be having segmentation faults when trying to read multiple bag files in a loop? Is it also a known issue?

All it is is the same code that extracts images from the bag file modified to loop over a list of bag files. Sometimes it runs on just 1 file before throwing a segmentation fault other times I get through 5-10 files. Seemingly random occurrence.

MartyG-RealSense, 2020-07-23:
There was a past case where a C++ user was experiencing segmentation errors when playing multiple bag files, so it may not be a problem that only affects Python.

https://github.com/IntelRealSense/librealsense/issues/2693

A RealSense support team member provided feedback in that discussion about a method of looping that worked for them whilst running a loop test.

https://github.com/IntelRealSense/librealsense/issues/2693#issuecomment-453725591

The mechanics of such loops and why segmentation faults may occur during them are outside of my programming knowledge, so I recommend posting a new issue on the GitHub asking about that particular subject if you would like to explore it further.

vidullan, 2020-07-27:
@MartyG-RealSense thanks for the link and all your help. I'll close this issue for now.

closed 2020-07-27 by vidullan

MartyG-RealSense, 2020-07-27:
Thanks very much for the update - good luck!