realsenseai/librealsense #6888: Pyrealsense2 RuntimeError: No device connected

sw14928, 2020-07-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515| 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC |
| SDK Version                            |  2.0                         | 
| Language                            |  python   | 
| Segment			|  Robot                       | 

### Issue Description
I am trying to run the basic examples to get my new L515 camera working on python. I am using Spyder IDE. I successfully pip install pyrealsense2, but every time I run the code I receive error: RuntimeError: No device connected

I checked the device manager and the camera is present. I checked the device works on the software it was shipped with. I checked the access was allowed. Can anyone help me with this error?

Many thanks

ev-mp, 2020-07-21:
@sw14928  hi,
> I successfully pip install pyrealsense2

Can you check that the pypi version is v2.35.2+?

> I checked the device works on the software it was shipped with

What software did you use, is it LRS v2.35.2/v2.36 ? If so - please share a screenshot for `rs-enumerate-devices` with the camera listed + Realsense-viewer's Camera info panel  in order to obtain initial camera properties.

sw14928, 2020-07-22:
Hi I am unable to check my pypi version? I receive the following when I attempt to check:

[image]

I am afraid I don't understand the second step - perhaps this is a problem of mine?

Many thanks!

RealSenseSupport, 2020-07-24:
Hi @sw14928 

Is it possible for you to download the RS Viewer (https://github.com/IntelRealSense/librealsense/releases/tag/v2.36.0) and provide the HW info that you are seeing when the L515 camera is seen after running RS Viewer?

You mentioned that you were able to see the camera enumerated in Device Manager and with the SW that came with it.  In debugging we are trying to figure out the details of the device, it's FW, etc shown in the RS Viewers info panel.

sw14928, 2020-07-27:
[image]

Hi @RealSenseSupport , this is the HW info I think? Does this help? Many thanks

ev-mp, 2020-07-27:
@sw14928 , what `pyrelasense2` version do you use? Please verify it is v2.35.2+.

sw14928, 2020-07-28:
For some reason my pip install only gave me 2.31.0.1235:

pyrealsense2 (2.36.0.2038)     - Python Wrapper for Intel Realsense SDK 2.0.
  INSTALLED: 2.31.0.1235
  LATEST:    2.36.0.2038

sw14928, 2020-07-28:
Ok this has solved my problem, I punched: 

pip install pyrealsense2==2.36.0.2038

into command line to fix

RealSenseSupport, 2020-07-31:
Glad to see it helped.  I'm assuming this can be closed out, correct?

closed 2020-08-18 by RealSenseSupport