realsenseai/librealsense #6890: Will d455 have slam capabilities similar to t265 ?

TheBricktop, 2020-07-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Raspberry Pi/ NVIDIA Jetson / Android  |
| SDK Version                            |  { 2}                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Smartphone/VR/AR}                          | 

### Issue Description
< question / feature request >

Since t265 is on the market for over a year now, and its a very unique and promising piece of a hardware, there is some demand in the area of mobile XR devices for more versatile device, like t265 with rgbd sensor. Right now to rely on realsense full spectrum of capabilities one have to strap two chunky sensors to a mobile device : t265 and any of the d400 (omitting d435i because two imu's on one device would be rather pointless) increasing the weight of the device even more. 
Considering that d455 specs are almost perfect for slam there could be a version with intel movidius inside.

MartyG-RealSense, 2020-07-23:
Hi @TheBricktop   Apologies for the delay in responding.  The RealSense D455's SLAM will be the same as the D435i model, and it will not perform SLAM in the way that the T265 does.

Intel test numerous different configurations of camera hardware. They do not comment on products that do not exist yet though.

TheBricktop, 2020-07-23:
So there will be slam, that's a good news, of only there was a way to use it outside Ros.

MartyG-RealSense, 2020-07-23:
Yes, the D455 has the IMU that the D435i has.

I recalled a D435i compatible SLAM project by MIT called **Kimera** and found the link to the details.

https://support.intelrealsense.com/hc/en-us/community/posts/360043774354-Highlighting-a-tutorial-for-Metric-Semantic-SLAM-with-Kimera-and-RealSense-D435i-

At 11 minutes 25 seconds into the YouTube video at the above link, it says that Kimera can be installed without ROS, though it takes longer (30 minutes) and more skill than an easy ROS based installation (5 minutes to begin compilation, 20 minutes to complete).

TheBricktop, 2020-07-24:
Problem with Ros is that it's mostly good for laboratory and experimental projects, for ar vr  especially now when mobile headsets take over its not very viable, correct me if I'm wrong but I don't see any way to port Ros based installation to for example unity Android.

MartyG-RealSense, 2020-07-24:
There is a free ROS wrapper for Unity created by Siemens, though I do not know of any cases where it has been used with RealSense.

https://assetstore.unity.com/packages/tools/physics/ros-107085
https://github.com/siemens/ros-sharp/wiki

closed 2020-08-05 by dorodnic

mpottinger, 2020-09-28:
@TheBricktop Yes please intel make a Depth + tracking camera with the capabilities of both the T265 and D400 series all in one device. 

I am sure I am not the only one who is really wishing for such a product.