realsenseai/librealsense #6893: Unstable depth values for each frame in D415

Dilip-Gurumurthy, 2020-07-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |  Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |             5.3.0                            | 
| Platform                                 | PC  |
| SDK Version                            |  2.29                     | 
| Language                            | Python 3                       | 
| Segment			|  Others                         | 

### Issue Description

I am measuring pixel depth with depth units set to 100um(i.e., 0.0001). But, the values keep varying for each frame. It varies by 100 - 1200 microns(.1 - 1.2 mm). I have tried using various combinations of presets as well as post-processing filters, some of them reduced the variation range, but still the values varied. My questions are as follows,

1. Why does the depth value fluctuate? Do you have any suggestions to make it return constant values?
2. How to measure depth in 10um(0.00001)? I know we can directly set it in the realsense viewer from FW 05.12.1, but that option is out of scope for me. Is there a way to set it from the python code itself?

Here's some additional information,
- Tested on surface: Metal(flat machined surface) | white paper | wood
- Light source: Sunlight(indoor) .
- I have measured depth from both live feed as well as bag files. In both cases, the values fluctuate.
- The distance is measured at close range(400-650 mm).
- At depth unit 1000um the values vary between 0 to 2 mm. But, for some filters, a few pixel depth values remain constant for a set of frames. However, I am interested only in depth units of 100um or less.
- Camera is fixed, facing down to a table. No moving entities in video. Well lit room.

Thanks.

MartyG-RealSense, 2020-07-22:
1.  If you have fluorescent lights such as ceiling strip lights in the location that the camera is used in then the lights may introduce noise into the image.  This is because the gas inside them flickers at frequencies that are hard to see with the human eye.  Using an FPS speed close to the operating frequency of such lights may help to reduce disruption.  For some lights it may be 30 FPS and for others it may be 60 FPS.

If the main light source is sunlight rather than artificial lights then this may be subject to more variation from moment to moment than a controlled artificial light source that produces a fixed, stable range and intensity of light that always projects over the same area.

If the surface being observed is a flat metallic surface though and the surface is reflective, there is the possibility that light may be reflecting off the metal's surface in some areas of it and making those areas more difficult for the camera to analyse for depth detail when the reflection is present.

Applying a physical filter such as a **linear polarizer** over the left and right camera lenses may help to reduce the negative effect of reflections, as described in **Section 4.4** of Intel's white-paper document about optical filters.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400

2.  You could set the depth units in Python like this:

**depth_sensor.set_option(rs.option.depth_units,0.0001)**

A script in the link below demonstrates an example of this:

https://github.com/IntelRealSense/librealsense/issues/2870

[image]

As far as I know though, the minimum depth scale is still **0.0001**.  I ran some tests with the depth unit option in the Viewer of SDK **2.36.0** just now.  Values less than 0.00005 became zero, and values greater than 0.00005 were rounded up automatically to 0.0001.

Dilip-Gurumurthy, 2020-07-29:
Thanks @MartyG-RealSense As always you are spot-on!
After going through your suggestion and following the links, I came to know that external factors like light source, depth surface, surface texture, etc., have a significant impact on depth quality. But, there are various options available to address them. However, due to delay in availability of required materials, I have yet to access the performance improvements by changing the external parameters. 

Regarding **depth_unit**, I guess the only way to set it to **0.00001** is by upgrading the FW to 5.12+

MartyG-RealSense, 2020-07-29:
I'm glad that I could be of help!  I look forward to your test results when you are able to access required materials.

I tested 0.00001 in the RealSense Viewer with the newest 5.12.6.0 firmware and it still reset the depth unit value to 0, so I don't think updating your firmware will make any difference in regard to being able to set a depth unit value lower than 0.0001.

MartyG-RealSense, 2020-08-08:
Hi @Dilip-Gurumurthy   Do you have any updates that you can provide yet, please?   Thanks!

Dilip-Gurumurthy, 2020-08-08:
Hi @MartyG-RealSense 

There is an improvement in the consistency of depth values after following your suggestion.
Here's what I did,
1. Stick polarizing film on both left and right stereo cameras. 
2. Enclose a non-flickering 9w LED bulb(flicker controlled- doesn't flicker even when capturing in slow motion from camera) with polarizing film in-order to have a polarized light source. 
3. Adjusted the films so that the reflections are completely eliminated.

I ran some experiments with varying presets and post-processing filters. Let's say, I take a hundred readings (with depth units set to 0.0001) of the same point(one pixel or more pixels averaged). Earlier the standard deviation was in the range of **50-300** microns. Now it has come down to **30-160** microns. But, for flat surfaces it is always less than **100** microns with or without filters. 

My concern is about the slight variations. The range of fluctuation has definitely reduced but small fluctuation is still an issue for me.

MartyG-RealSense, 2020-08-08:
For flat surfaces (especially ones with little or no texture), you could try having the IR Emitter enabled so that the projector protects a pattern of semi-random dots onto surfaces in the scene.  The camera can use these dots as a texture source to analyse them for depth detail.

If the strong lighting in the room is not illuminating the flat surfaces strongly, the dot pattern projection may make a positive distance to the camera's ability to read detail from those surfaces.