realsenseai/librealsense #6894: can l515 be used for finger tracking?

AndreV84, 2020-07-22:
any information on the topic?

RealSenseSupport, 2020-08-13:
Hi @AndreV84 

From a quality standpoint, L515 can do finger tracking well.  Intel does not provide a middleware package that does finger tracking for this camera.  We hope developers in the community or SW vendors can step in and produce such a middleware.

I'd like to point you to a blog that was done some time back on our site specifically talking about hand and gesture tracking for your reference:
https://www.intelrealsense.com/hand-tracking-overview/

AndreV84, 2020-08-13:
Thank you for the update!
the blog article neither provides any software solution to implement the idea defined in the blog;
not sure why third party developers should supply software for project defined by Intel with Intel's hardware in Intes's marketing blog;
It would make sense if Intel provided software [ example] to implement tracking;
I know few open-source solutions;
Including openpose by Carnegie Melon University group
but they are generic and hardly can benefit in particular from intels hardware, in my opinion

AndreV84, 2020-08-16:
How to use l515 with open3d?
like in https://forums.developer.nvidia.com/t/jetscan-portable-and-instant-rgb-d-3d-scanner/116476/3 ?

RealSenseSupport, 2020-09-11:
Is there anything further needed on this one?

*If we don’t hear from you in 7 days, this issue will be closed.

AndreV84, 2020-09-12:
it seems that l515 as for now is not supported by open3d , while d400 is supported
is l515 integration planned?

RealSenseSupport, 2020-09-22:
Nothing planned from Intel side to add support for L515

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-10-05 by RealSenseSupport

AndreV84, 2020-10-05:
any help with getting L515 activated in warehouse beta software package? It won't get through
Thanks