realsenseai/librealsense #6895: Error:Frame did not arrived with in 5000 ..with my xavier NX

sohail759, 2020-07-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { Intel 534i} | 
| Firmware Version                   | (using python library pyrealsence2 ) | 
| Operating System & Version |   { jetpack }                                      | 
| Platform                                 |  NVIDIA Jetson Xavier NX   |

I am using realsence camera with nvidia jetson xavier NX ,I am using pipeline.start(config) to start the pipeline and pipeline.stop() function  to stop the pipeline..my camera works fine for the 1st time but when I rerun the python code without reboot my xavier the error pop up  Error:Frame did not arrived with in 5000...can anyone help me to resolve this issue.

MartyG-RealSense, 2020-07-22:
Hi @sohail759   May I confirm which RealSense camera model you are using, please?  Is it the **D435i** ?  Thank you very much.

sohail759, 2020-07-22:
Yes 435i

MartyG-RealSense, 2020-07-22:
Thank you very much.  If you are able to use the **RealSense Viewer** program, would it be possible please to go to the **More** option at the top of the Viewer's options side-panel and update the camera's firmware driver by selecting **Install Recommended Firmware** from the More option's drop-down menu?

sohail759, 2020-07-22:
Thanks for replying so quickly..is this work with python?

MartyG-RealSense, 2020-07-22:
If you are using an Xavier, you likely have librealsense installed on the Ubuntu OS, yes?  Then you would launch the RealSense Viewer from Ubuntu.

closed 2020-08-08 by MartyG-RealSense