realsenseai/librealsense #6897: Build error with flag BUILD_NETWORK_DEVICE=ON

podshivalov-ilya, 2020-07-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |   Linux (Manjaro)  | 
| Kernel Version (Linux Only)    |  5.7.9                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.35.2                           | 
| Language                            |  C                          | 
| Segment			|  others                          | 

### Issue Description
In attempt to build SDK from tag v2.35.2 with the following configuration
`
cmake .. \
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DCMAKE_INSTALL_LIBDIR=lib \
    -DCMAKE_INSTALL_SBINDIR=bin \
    -DCMAKE_INSTALL_SYSCONFDIR=/etc \
    -DCMAKE_BUILD_TYPE=Release \
    -DBUILD_SHARED_LIBS=on \
    -DBUILD_WITH_STATIC_CRT=off \
    -DBUILD_WITH_OPENMP=on \
    -DBUILD_EXAMPLES=true \
    -DBUILD_WITH_TM2=true \
    -DBUILD_NETWORK_DEVICE=ON
cmake --build . -- -j8
`
I've received a linker error during building SDK:
`[ 39%] Linking CXX executable rs-server
c++: error: ;-pthread: No such file or directory
make[2]: *** [tools/rs-server/CMakeFiles/rs-server.dir/build.make:3167: tools/rs-server/rs-server] Error 1
make[1]: *** [CMakeFiles/Makefile2:2054: tools/rs-server/CMakeFiles/rs-server.dir/all] Error 2
`
I've resolved this issue with 
[patch](https://github.com/IntelRealSense/librealsense/files/4961138/cmake_pthread.txt) but I don't know is it appropriate fix. Moreover, I have checked that patch doesn't break linkage to libpthread with ldd rs-server

dorodnic, 2020-07-23:
Hi @podshivalov-ilya 
I think this has something to do with the version of CMake / GCC installed on your system. Other projects also seem to encounter this problem ([OpenIGTLink/issues/218](https://github.com/openigtlink/OpenIGTLink/issues/218))
If the code compiles on your system without this term everything should work Ok.

podshivalov-ilya, 2020-07-23:
Thank you @dorodnic. I use gcc 10.1 and cmake 3.17.3 and I haven't tried to build SDK with older versions. Does this [set](https://github.com/IntelRealSense/librealsense/blob/master/tools/rs-server/CMakeLists.txt#L19) do the same as [THREADS_PREFER_PTHREAD_FLAG](https://cmake.org/cmake/help/latest/module/FindThreads.html#variable:THREADS_PREFER_PTHREAD_FLAG)?

dorodnic, 2020-07-23:
In other SDK examples we are using - 
```
set(THREADS_PREFER_PTHREAD_FLAG ON)
find_package(Threads REQUIRED)
```
and use `Threads::Threads`. This seem to be better than directly setting linker flag. I think the author of the server should have used this method as well.

RealSenseSupport, 2020-07-27:
Hi @podshivalov-ilya Do you need further help on this? If we don't hear from you in 7 days, this issue will be closed.

podshivalov-ilya, 2020-07-28:
Hello. No, I don't need any help. Thanks

closed 2020-07-28 by podshivalov-ilya

3dsf, 2020-11-01:
I was getting this error too, and in the end it was due to my `$LD_LIBRARY_PATH` being wrong as it did not include `/usr/local/lib64/` in my case