realsenseai/librealsense #6898: RealSense device no detected in Android

jeronimoagullo, 2020-07-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.06.00 5.12.6.0| 
| Operating System & Version |   Win (10)  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | Android 9 in Snapdragon 845 |
| SDK Version                            |   28                        | 
| Language                            |  C#/unity                       | 
| Segment			|  Smartphone/AR                       | 

### Issue Description
I am trying to use the RealSense D435i in an Snapdragon development board running Android 9 using Unity. **The issue is that my app doesn't recognize the RealSense camera**.

I have already had success building and running the camera example using Android Studio. Now, I am trying to build the examples of the RealSense wrapper in Unity for an Android 9 device. However, I didn't have success.

### Followed steps
1. I have downloaded the Unity package and import in an Unity project with version 2019.3.11f1
2. I have compiled the library "librealsense.aar" in Android Studio and place in the Unity folder "Assets/RealSenseSDK2.0/Plugins" with the name "librealsense"
3. The project works and recognise the camera in the Unity editor in Windows 10.
4. **The project builds successfully and the app is launched. However, the RealSense camera is not detected**.
I am trying to build the Scene Assets/RealSenseSDK2.0/Scenes/StartHere 
What am I missing? 

[image]

### Logcat of the app
I attach the logcat when running the app.

--------- beginning of main
07-22 15:30:37.795 18178 18178 I .RealSenseTest: Late-enabling -Xcheck:jni
07-22 15:30:37.967 18178 18178 I .RealSenseTest: The ClassLoaderContext is a special shared library.
07-22 15:30:38.045 18178 18178 I Perf    : Connecting to perf service.
07-22 15:30:41.436 18178 18178 W ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@6bc2236
07-22 15:30:41.498 18178 18178 D Unity   : CommandLine:  
07-22 15:30:41.501 18178 18178 D OpenGLRenderer: Skia GL Pipeline
07-22 15:30:41.517 18178 18178 I Unity   : onResume
07-22 15:30:41.564 18178 18236 D Unity   : SetWindow 0 0xecdcd808
07-22 15:30:41.564 18178 18236 D Unity   : SetWindow 0 0xecdcd808
07-22 15:30:41.578 18178 18178 I Unity   : windowFocusChanged: true
--------- beginning of system
07-22 15:30:41.600 18178 18236 E SchedPolicy: set_timerslack_ns write failed: Operation not permitted
07-22 15:30:41.633 18178 18236 D Unity   : Enabling Unity systrace
07-22 15:30:41.637 18178 18236 W .RealSenseTest: Accessing hidden method Ljava/lang/invoke/MethodHandles$Lookup;-><init>(Ljava/lang/Class;I)V (light greylist, reflection)
07-22 15:30:41.646 18178 18236 D Unity   : [VFS] Mount /data/app/com.DefaultCompany.RealSenseTests-cXBHuHm9CmUk4gycKzOgpQ==/base.apk
07-22 15:30:41.648 18178 18236 D Unity   : Loading player data from /data/app/com.DefaultCompany.RealSenseTests-cXBHuHm9CmUk4gycKzOgpQ==/base.apk/assets/bin/Data/data.unity3d
07-22 15:30:41.658 18178 18236 I Unity   : SystemInfo CPU = ARM64 FP ASIMD AES, Cores = 8, Memory = 3674mb
07-22 15:30:41.658 18178 18236 I Unity   : SystemInfo ARM big.LITTLE configuration: 4 big (mask: 0xf0), 4 little (mask: 0xf)
07-22 15:30:41.659 18178 18236 I Unity   : ApplicationInfo com.DefaultCompany.RealSenseTests version 0.1 build 0fce3bfc-ab13-4db6-9b21-b0c1d7c92994
07-22 15:30:41.659 18178 18236 I Unity   : Built from '2019.3/staging' branch, Version '2019.3.11f1 (ceef2d848e70)', Build type 'Development', Scripting Backend 'mono', CPU 'armeabi-v7a', Stripping 'Disabled'
07-22 15:30:41.666 18178 18236 D Unity   : Mono path[0] = '/data/app/com.DefaultCompany.RealSenseTests-cXBHuHm9CmUk4gycKzOgpQ==/base.apk/assets/bin/Data/Managed'
07-22 15:30:41.666 18178 18236 D Unity   : Mono config path = 'assets/bin/Data/Managed/etc'
07-22 15:30:41.666 18178 18236 D Unity   : PlayerConnection initialized from /data/app/com.DefaultCompany.RealSenseTests-cXBHuHm9CmUk4gycKzOgpQ==/base.apk/assets/bin/Data (debug = 0)
07-22 15:30:41.667 18178 18236 D Unity   : PlayerConnection initialized network socket : 0.0.0.0 55351
07-22 15:30:41.667 18178 18236 D Unity   : PlayerConnection initialized unix socket : Unity-com.DefaultCompany.RealSenseTests
07-22 15:30:41.668 18178 18236 D Unity   : Multi-casting "[IP] 192.168.0.27 [Port] 55351 [Flags] 2 [Guid] 363998825 [EditorId] 3978031801 [Version] 1048832 [Id] AndroidPlayer(QUALCOMM_SDM845_for_arm64@192.168.0.27) [Debug] 0 [PackageName] AndroidPlayer [ProjectName] RealSenseTests" to [225.0.0.222:54997]...
07-22 15:30:41.668 18178 18236 D Unity   : Started listening to [0.0.0.0:55351]
07-22 15:30:41.740 18178 18236 D Unity   : [EGL] Attaching window :0xecdcd808
07-22 15:30:41.741 18178 18236 D Unity   : InitializeScriptEngine OK (0xed20ff00)
07-22 15:30:41.741 18178 18236 D Unity   : PlayerConnection already initialized - listening to [0.0.0.0:55351]
07-22 15:30:41.763 18178 18236 D Unity   : Loading player data from assets/bin/Data/data.unity3d
07-22 15:30:41.764 18178 18236 D Unity   : PlayerInitEngineNoGraphics OK
07-22 15:30:41.764 18178 18236 D Unity   : AndroidGraphics::Startup window =  0xecdcd808
07-22 15:30:41.764 18178 18236 D Unity   : [EGL] Attaching window :0xecdcd808
07-22 15:30:41.768 18178 18236 D Unity   : [Subsystems] Discovering subsystems at path assets/bin/Data/UnitySubsystems
07-22 15:30:41.774 18178 18236 D vulkan  : searching for layers in '/data/app/com.DefaultCompany.RealSenseTests-cXBHuHm9CmUk4gycKzOgpQ==/lib/arm'
07-22 15:30:41.774 18178 18236 D vulkan  : searching for layers in '/data/app/com.DefaultCompany.RealSenseTests-cXBHuHm9CmUk4gycKzOgpQ==/base.apk!/lib/armeabi-v7a'
07-22 15:30:41.774 18178 18236 D Unity   : [Vulkan init] extensions: count=10
[... 149 lines cut ...]
07-22 15:30:44.586 18178 18236 E Unity   : [./Runtime/Serialize/SerializedFile.cpp line 211] 
07-22 15:30:44.586 18178 18236 E Unity   : (Filename: /Users/builduser/buildslave/unity/build/Modules/AssetBundle/Managed/AssetBundle.bindings.cs Line: 87)
07-22 15:30:44.586 18178 18236 E Unity   : 
07-22 15:30:44.587 18178 18236 E Unity   : The AssetBundle 'sample_scenes' can't be loaded because it was not built with the right version or build target.
07-22 15:30:44.587 18178 18236 E Unity   : UnityEngine.AssetBundle:LoadFromFile_Internal(String, UInt32, UInt64)
07-22 15:30:44.587 18178 18236 E Unity   : UnityEngine.AssetBundle:LoadFromFile(String) (at /Users/builduser/buildslave/unity/build/Modules/AssetBundle/Managed/AssetBundle.bindings.cs:87)
07-22 15:30:44.587 18178 18236 E Unity   : SceneLoader:Awake() (at C:\workSpace\Unity-workspace\RealSenseTests\RealSenseTests\Assets\RealSenseSDK2.0\Misc\Utils\SceneLoader.cs:16)
07-22 15:30:44.587 18178 18236 E Unity   : 
07-22 15:30:44.587 18178 18236 E Unity   : [./Modules/AssetBundle/Public/AssetBundleLoadFromAsyncOperation.cpp line 445] 
07-22 15:30:44.587 18178 18236 E Unity   : (Filename: /Users/builduser/buildslave/unity/build/Modules/AssetBundle/Managed/AssetBundle.bindings.cs Line: 87)
07-22 15:30:44.587 18178 18236 E Unity   : 
07-22 15:30:44.621 18178 18236 D Unity   : Choreographer available: Enabling VSYNC timing
07-22 15:30:44.697 18178 18236 I librs   : Found 0 RealSense devices (mask 254)
07-22 15:30:44.959 18178 18286 D Unity   : Setting up 1 worker threads for Enlighten.
07-22 15:30:44.959 18178 18287 D Unity   :   Thread -> id: ffffffffc6247970 -> priority: 1 
07-22 15:30:45.003 18178 18292 I librs   : Found 0 RealSense devices (mask 254)
07-22 15:30:45.004 18178 18292 I librs   : Found 0 RealSense devices (mask 255)
07-22 15:30:45.004 18178 18290 I librs   : Found 0 RealSense devices (mask 254)
07-22 15:30:45.004 18178 18290 I librs   : Found 0 RealSense devices (mask 255)
07-22 15:30:45.004 18178 18291 I librs   : Found 0 RealSense devices (mask 254)
07-22 15:30:45.004 18178 18291 I librs   : Found 0 RealSense devices (mask 255)
07-22 15:30:45.004 18178 18289 I librs   : Found 0 RealSense devices (mask 254)
07-22 15:30:45.004 18178 18289 I librs   : Found 0 RealSense devices (mask 255)

Thank you in advance

MartyG-RealSense, 2020-07-22:
Hi @jeronimoagullo   In a couple of previous cases where RealSense on Android was being integrated with Unity, the RealSense users in those cases got it working by adding a request for the camera permission and the RsContext init.

https://github.com/IntelRealSense/librealsense/issues/6526
https://github.com/IntelRealSense/librealsense/issues/4155

The camera permission script is at the link below:

https://github.com/IntelRealSense/librealsense/issues/4155#issuecomment-501208611

An additional step that may sometimes need to be taken is to use a value of '27' for **targetSdkVersion**.  This is usually only needed for Android 10 devices, but I post it here in case you have further problems and wish to try it with your Snapdragon board.

TheBricktop, 2020-07-22:
I think there was some problems with vulcan and il2cpp also, ive managed to build project to android using this configuration https://github.com/GeorgeAdamon/quest-realsense .

jeronimoagullo, 2020-08-03:
Hello @MartyG-RealSense and @TheBricktop 

It is almost working. The camera is now recognised. However there is some errors in the Unity editor. Althought the apk is built, there is the following errors:

> Component GUI Layer in Main Camera for Scene Assets/RealSenseSDK2.0/Scenes/Samples/PointCloudProcessingBlocks.unity is no longer available.
> It will be removed after you edit this GameObject and save the Scene.
> UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)

And it happens the same for the Scenes: AlignmentSample, PointCloudDepthAndColor, TexturesDepthAndColor and TexturesDepthAndInfrared.

If I click in the error, the script which asks for the Android permission is opened. I imagine that this script is the cause of the errors. I have created a new empty GameObject in the StartHere scene, then I have added the permission script to it. Have I missed something?

Thank you in advance

MartyG-RealSense, 2020-08-03:
@jeronimoagullo   I have seen this error myself during general Unity use, though not during RealSense related use.  I believe that it occurred because I installed something that an object in the Unity project depended upon and then deleted that asset and the project could no longer access that component, and so auto-removed it.

If there are missing assets, running the RealSense Unity installer file **realsense.unitypackage ** again may re-import them into the project.

MartyG-RealSense, 2020-08-15:
@jeronimoagullo   Do you still require assistance with this case, please?   Thanks!

closed 2020-08-27 by MartyG-RealSense