realsenseai/librealsense #6899: Re-starting the color stream crashes the camera

jurc192, 2020-07-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 5.12.06 | 
| Operating System & Version | Linux Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  5.4                              | 
| Platform                                 | PC  |
| SDK Version                            | 2.36.0 | 
| Language                            |  python                          | 
| Segment			|  others                           | 

### Issue Description
Using the RealSense viewer and my D435i camera, the RGB stream only works when enabling it the first time. Stopping and re-enabling the stream does not work, 'no frames arrived' error message is shown. Plugging the camera's USB out and in again helps.

This was tested on two Linux computers (Ubuntu and Arch), latest versions, using compiled realsense with RSUSB=1 testing with RealSense Viewer. **HOWEVER, this problem does not appear on every USB ports/controllers:**
- using the port driven by `Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)` host controller works
- using the port driven by `ASMedia Technology Inc. ASM1142 USB 3.1 Host Controller` host controller does not work (tested on two different computers)

Could the issue be hardware related? How come the stream work the first time?

Here are the logs when using the ASMedia technology host controller, and re-starting the RGB stream using RealSense Viewer (Ubuntu 20.04):
[singlecam_dmesg.log](https://github.com/IntelRealSense/librealsense/files/4962402/bigfoot_singlecam_dmesg.log)
```
...
[ 1370.391612] hid-sensor-hub 0003:8086:0B3A.0008: No report with id 0xffffffff found
[ 1429.940193] xhci_hcd 0000:09:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
```

[singlecam_realsense.log](https://github.com/IntelRealSense/librealsense/files/4962411/bigfoot_singlecam_realsense.log)

MartyG-RealSense, 2020-07-23:
Hi @jurc192   I note that you have updated the camera to the most recent firmware and that you have built Ubuntu 20.04 with the RSUSB installation method.  So that covers the first two suggestions that I would have recommended trying.

It does seem likely though that it is an issue related to a specific USB port rather than a firmware or SDK issue.  

If you know how to access the **BIOS** settings when your computer is booting up and you feel confident in making changes to the configuration settings, could you check if your computer's BIOS has an option called **xHCI** please?  If it does, set it to **Enabled**(not Auto or Smart Auto).

jurc192, 2020-07-23:
Thanks for your advice @MartyG-RealSense !
I will try this and report back next week :)

jurc192, 2020-08-01:
No, none of my two computers has anything related to xHCI in the BIOS... I tried changing various settings related to USB but nothing changes.

MartyG-RealSense, 2020-08-01:
Apologies for the delay in responding further, as I was carefully considering your case.

It would be useful to know the differences between your **Intel xHCPI  controller** port that the RGB stream works with, and the **ASM1142 USB 3.1 Host Controller** port that it doesn't work with.  Are they all USB ports built into the computer, or is one port type on the computer and the other type on a USB hub or USB expansion card?  Thanks for your patience!

jurc192, 2020-08-02:
So I have two computers, my laptop and a workstation. 
### Laptop
- It has 4 exposed USB ports: 2x USB2, 1x USB3 type A, 1x USB3 type C
- `lspci -v -t` output:
```
-[0000:00]-+-00.0  Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers
           +-02.0  Intel Corporation UHD Graphics 620
           +-14.0  Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller
           +-14.2  Intel Corporation Sunrise Point-LP Thermal subsystem
           +-16.0  Intel Corporation Sunrise Point-LP CSME HECI #1
           +-17.0  Intel Corporation Sunrise Point-LP SATA Controller [AHCI mode]
           +-1c.0-[01]--+-00.0  Realtek Semiconductor Co., Ltd. RTL8411B PCI Express Card Reader
           |            \-00.1  Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
           +-1c.3-[02]----00.0  Intel Corporation Wireless 8265 / 8275
           +-1c.4-[03]----00.0  ASMedia Technology Inc. ASM1142 USB 3.1 Host Controller
           +-1f.0  Intel Corporation Sunrise Point LPC Controller/eSPI Controller
           +-1f.2  Intel Corporation Sunrise Point-LP PMC
           +-1f.3  Intel Corporation Sunrise Point-LP HD Audio
           \-1f.4  Intel Corporation Sunrise Point-LP SMBus
```
Here we see that there are two USB Host controllers (14.0 and 1c.04-00.0)

- info about the **Intel USB controller** (using `sudo lspci -s 14.0 -v`)
```
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21) (prog-if 30 [XHCI])
	Subsystem: CLEVO/KAPOK Computer Device 7a01
	Flags: medium devsel, IRQ 128
	Memory at df310000 (64-bit, non-prefetchable) [size=64K]
	Capabilities: [70] Power Management version 2
	Capabilities: [80] MSI: Enable+ Count=1/8 Maskable- 64bit+
	Kernel driver in use: xhci_hcd
	Kernel modules: xhci_pci
```

- info about the **ASMedia USB controller** (using `sudo lspci -s 3:00 -v`)
```
03:00.0 USB controller: ASMedia Technology Inc. ASM1142 USB 3.1 Host Controller (prog-if 30 [XHCI])
	Subsystem: ASMedia Technology Inc. ASM1142 USB 3.1 Host Controller
	Flags: bus master, fast devsel, latency 0, IRQ 16
	Memory at df000000 (64-bit, non-prefetchable) [size=32K]
	Capabilities: [50] MSI: Enable- Count=1/8 Maskable- 64bit+
	Capabilities: [68] MSI-X: Enable+ Count=8 Masked-
	Capabilities: [78] Power Management version 3
	Capabilities: [80] Express Endpoint, MSI 00
	Capabilities: [100] Virtual Channel
	Capabilities: [200] Advanced Error Reporting
	Capabilities: [280] Secondary PCI Express
	Capabilities: [300] Latency Tolerance Reporting
	Kernel driver in use: xhci_hcd
	Kernel modules: xhci_pci
```

**The camera works on all USB ports, except the USB3 type C one (which is the only USB port controlled by the ASMedia Host Controller)**.
(Besides 'no frames' in realsense, also this message appears in `dmesg` when restarting the stream:
`xhci_hcd 0000:03:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.`

### Workstation
- Has a USB3 PCI expansion card ([this one](https://www.startech.com/Cards-Adapters/USB-3.0/Cards/4-port-usb-3-1-card~PEXUS313AC2V)) with 3x USB3 type A ports and 1x USB3 type C port:
`ASMedia Technology Inc. ASM1142 USB 3.1 Host Controller` (**identical to the laptop one**, mentioned above)
- Has multiple USB2 ports, driven by `Intel Corporation 82801JI (ICH10 Family) USB UHCI Controller` 
- I don't have computer here at the moment to show `lspci -t -v` but basically, the problem appears on all USB3 ports (driven by ASMedia). However, restarting RGB streams on the USB2 ports (driven by Intel controller) works okay .

That is why I think there might be correlation between this, since none of the ASMedia controllers can handle restarting of RGB streams, but all Intel controllers can (regardless of USB version and connector type and kernel version (Laptop is Arch linux latest, Workstation is latest Ubuntu)...

MartyG-RealSense, 2020-08-03:
Sincere apologies for the delay in responding further.

May I ask if there is a reason why you cannot use the camera in the 1 USB3 port on the laptop if it works in that (e.g it is occupied by another device that needs USB3).  If you have a device in the USB3 port that is only there because you ran out of empty USB2 ports, an approach that I use myself is to connect an unpowered USB2 four-port hub with a short lead to one of the USB2 ports on the laptop.  

Because an unpowered hub draws its power from the computer, the laptop remains mobile (unlike a mains powered hub where you are tethered to a mains power socket).  The unpowered hub can run all the low-power devices that only need USB2 to operate, leaving the USB3 port vacant for a device that benefits from USB3 (such as a 400 Series camera).

A similar strategy could be used with the workstation to avoid using the camera in the USB-C Asmedia port.

jurc192, 2020-08-03:
I am using RealSense cameras to build a 4D scanner for human body (3D scanning in time, motion), so the goal is to develop
dedicated embedded system which will handle multiple RealSense cameras. We bought a high-end USB3 PCI expansion card, because we are aware of heavy bandwidth requirements, but there are more basic problems we want to solve first before going to multiple cameras and pushing the limits of our hardware.
I opened this bugreport/issue because I noticed that a very basic usage of camera+SDK causes problems and I hope that we can find the bug together so we make RealSense SDK better and easier to use for everyone. My motivation for opening this kind of bugreports is to make RealSense SDK more robust and Linux-friendly, so more people can invent cool stuff with these cameras.
TL;DR hub is not an option, we need more bandwidth + we want to make D435i work on any USB3 port/controller

I am ready to investigate and learn further, but having some guidance from the developers would be useful.

Finally, there is no need to apologise for the delay in responding. I think that your support here on GitHub is amazing and one of fastest and I really appreciate the amount of hard work you guys are putting in this! Keep up the good work and take your time, also time for yourself ;) :)

MartyG-RealSense, 2020-08-03:
Thanks so much for your kind words!   :)

In regard to the PCI-e expansion card, it is not a guarantee of RealSense camera stability.  This is because it draws its power from the computer's power supply unit, and so it may be vulnerable to the same power supply instabilities that can come with plugging the camera directly into a computer's built-in USB port or using an unpowered hub (which also draws its power from the computer PSU).  Mains-powered hubs can provide greater USB stability for the camera, though it is not an option in this particular project if you cannot use any hub at all.

Having said that about PCI-e, using industrial-grade equipment can make a positive difference.  As well as high-grade USB hardware, you could also consider industrial-grade USB cables if you are not using them already.  These are used in critical applications where you cannot tolerate dropping of frames, such as hospital medical scanners.  

Intel's officially recommended supplier of USB cables for the RealSense 400 Series cameras, **Newnex**, is an industrial-grade cable supplier.

https://www.newnex.com/realsense-3d-camera-connectivity.php

You can also reduce bandwidth by using depth image compression, as described in Intel's white-paper document on the subject.

https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras

The RealSense developers introduced improvements to multicam support in SDK version **2.35.2**, though it addresses problems that may occur in the rs2::pipeline and may not solve problems that occur because of specific hardware such as a particular make of USB hub.

If you also need higher FPS speed, you can use intel's External Synchronization (genlock) feature to connect multiple cameras together in a trigger sequence to amplify FPS.  For example, 10 cameras at 848x480 and 90 FPS for a total FPS of 900 FPS, or ten cameras at 848x100 and 300 FPS for 3000 FPS.  Intel has a white paper on this feature too.

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras

jurc192, 2020-08-11:
Thanks for the tips. I would like to keep this thread related to a single-camera operation, since I believe it's more important (if single camera doesn't work, multiple cameras will probably not work either). Let us assume cable and PCI card are ok for now.

After some more investigating, I noticed that:
- Enabling RGB stream on ASMedia USB host controllers only works the first time (second time=> no frames arrived); as said before
- **Rebinding the `xhci_hcd` drivers fixes the issue (a.k.a. software resetting the complete USB host controller).**

So if I rebind the USB drivers everytime before enabling the stream - it will always work (according to my tests).
This problem does not appear when using the Intel Host controller.

So this is how my reset_usb.sh script looks like:
```
#!/bin/bash
# Taken from https://askubuntu.com/a/290519/925619 
# Resets all devices driven by the xhci driver (USB3 devices, overkill, but simple)
for i in /sys/bus/pci/drivers/xhci_hcd/*:*; do
  [ -e "$i" ] || continue
  echo "${i##*/}" > "${i%/*}/unbind"
  echo "${i##*/}" > "${i%/*}/bind"
done
sleep 0.5
```

I execute the script everytime before creating the `pipeline` and `config` and starting the pipeline.

MartyG-RealSense, 2020-08-11:
Thanks so much for sharing your scripting solution!

If you still have a remaining problem that needs solving, can you explain what it is please?  Thank!

jurc192, 2020-08-11:
Well this is just a nasty hack / workaround which is ok for now, but I would like to (help) finding the **real issue** not workarounds.
I would be happy if a developer could comment on this, but I understand if it's not the first priority (even though this bug is really annoying). 

There is another thing I noticed: In scenario with multiple cameras, there were problems if you started the pipelines without  adding some delays in between. Like suggested [here](https://github.com/IntelRealSense/librealsense/issues/6343#issuecomment-623332094) (but I saw this suggestion in multiple places). **However, rebinding the xhci_hcd drivers as shown above allows starting each camera (once) without any delays in between** and from my experience it always works - saves lots of time and patience.

I am planning to dive into the SDK / context implementation and video4linux drivers, but some guidance would be nice, if somebody could point me to right direction - which things to focus on etc.

MartyG-RealSense, 2020-08-19:
Apologies for the lack of further responses on your case.  Another RealSense user has been having problems with launching multiple cameras and made a detailed testing report.  It might provide some useful insight.

https://github.com/IntelRealSense/realsense-ros/issues/1187#issuecomment-675786918

MartyG-RealSense, 2020-08-27:
Hi @jurc192   Do you still require assistance with this case, please?  Thanks!

jurc192, 2020-08-27:
I do not require extra assistance at the moment, but I do believe that the issue is not yet resolved and should be looked into by a developer. I will give an update if I figure out anything new, until then we can close the issue and thanks for your support!

closed 2020-08-27 by jurc192