realsenseai/librealsense #6902: How to load .json file for L515 camera in python?

zeze1111, 2020-07-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | (2.35.2) | 
| Operating System & Version |   Linux (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python                         | 
| Segment			|                           | 

### Issue Description
In python interface, I can load the .json file to change parameters of the D400 camera. But it is rs400_advanced_mode, and it doesn't work for L515 camera. And I don't see any instruction to guide how to use extension.serializable in Python. 
So how can I load the .json file extrated from realsense-viewer for a L515 camera within Python???

RealSenseSupport, 2020-08-13:
Hi @zeze1111 

Thanks for your inquiry.  We are investigating this.

aidonchuk, 2020-12-13:
Hi! Any updates on L515?

maloel, 2021-01-07:
@aidonchuk 
I don't know why nobody responded here but...
The proper way to load a .json file is by using `rs2::serializable_device`, the base class for `rs400::advanced_mode`.
That said, the interface is not yet exposed in Python. It will only be exposed in the next release.
You can cherry pick the commit that adds the interface, though, #7895.
Another note: if the .json has a Visual Preset of "Default" then you may run into exceptions when trying to load it. That has still not been fixed but will be available in the next release.

eddiz, 2021-01-20:
@maloel 
loading the json file for L515, created with RealSense Viewer using `rs2::serializable_device` raises an exception for me. Same code works well for D455 and `rs400::advanced_mode` Is this a known issue and was fixed in #7895 as well?

System: Win 10, SDK 2.41.0. L515

RealSenseSupport, 2021-01-29:
@eddiz 

There is still some work being done on this.  Will notify this thread when we know more.

RealSenseSupport, 2021-02-10:
Here is a PR that has been merged into development:  https://github.com/IntelRealSense/librealsense/pull/8002

autimator, 2021-02-18:
Hello, I just tested this code with the example from https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-rs400-advanced-mode-example.py

But the product Id of my lidar (0B64) is not in the list ('DS5_product_ids').
When I remove that check I get the following error:
""RuntimeError: null pointer passed for argument "dev"""

On this line:  while not advnc_mode.is_enabled():

Does this mean the advanced mode is still not available for the L515?

eddiz, 2021-02-18:
@autimator: The advanced mode is limited to the D400 device class. But loading the json files is facilitated by the parent `rs2::serializable_device`. Did this resolved your issues? I had no time to test the updated version so far.

autimator, 2021-02-18:
@eddiz  thanks for the tip! I just tested this code on my L515 (no errors):

 ```def find_device_that_supports_advanced_mode(config):
    ctx = rs.context()  
    devices = ctx.query_devices()
    for dev in devices:       
    	print("Found device:", dev.get_info(rs.camera_info.name))
        return dev 

         #load json
        dev = find_device_that_supports_advanced_mode(config) 
        ser_dev = rs.serializable_device(dev)  
        with open(jsonfile, 'r') as file:
            json = file.read().strip()       
            ser_dev.load_json(json )
            print("loaded json")

RealSenseSupport, 2021-03-01:
Are the issues with the json file taken care of now?

If we don’t hear from you in 7 days, this issue will be closed.

GreatScherzo, 2021-03-09:
Sorry for the abrupt comment.

The issue seems to not be resolved yet. The code that @autimator made above did load the json string, but the output image stream from the L515 doesn't seems to reflect the parameters from the string.

Heres a code that I made up until now in attempt to load the json file and stream L515

```
import pyrealsense2 as rs
import numpy as np
import cv2
import json

#%%
DS5_product_ids = ["0AD1", "0AD2", "0AD3", "0AD4", "0AD5", "0AF6", "0AFE", "0AFF", "0B00", "0B01", "0B03", "0B07", "0B3A", "0B5C", "0B64"]

def find_device_json_input_interface() :
    ctx = rs.context()
    ds5_dev = rs.device()
    devices = ctx.query_devices();
    for dev in devices:
        if dev.supports(rs.camera_info.product_id) and str(dev.get_info(rs.camera_info.product_id)) in DS5_product_ids:
            if dev.supports(rs.camera_info.name):
                print("Found device", dev.get_info(rs.camera_info.name))
            return dev
    raise Exception("No product line device that has json input interface")

jsonDict = json.load(open("LCustom.json"))
jsonString= str(jsonDict).replace("'", '\"')

#%%
# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()

try:
    dev = find_device_json_input_interface()
    ser_dev = rs.serializable_device(dev)  
    ser_dev.load_json(jsonString)
    print("loaded json")

except Exception as e:
    print(e)
    pass
#%%

# Get device product line for setting a supporting resolution
pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

if device_product_line == 'L500':
    config.enable_stream(rs.stream.color, 960, 540, rs.format.bgr8, 30)
else:
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Start streaming
pipeline.start(config)

#後処理フィルタブロック定義
decimation = rs.decimation_filter()
# decimation.set_option(rs.option.filter_magnitude, 4)

depth_to_disparity = rs.disparity_transform(False)

spatial = rs.spatial_filter()
# spatial.set_option(rs.option.filter_magnitude, 5)
# spatial.set_option(rs.option.filter_smooth_alpha, 1)
[... 69 lines cut ...]
# 
# =============================================================================

        if depth_colormap_dim != Postframe_colormap_dim:
            resized_Postframe_image = cv2.resize(Postframe_colormap, dsize=(depth_colormap_dim[1], depth_colormap_dim[0]), interpolation=cv2.INTER_AREA)
            images = np.hstack((resized_Postframe_image, depth_colormap))
        else:
            images = np.hstack((Postframe_colormap, depth_colormap))            

            
            
        # Show images
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', images)
        cv2.waitKey(1)

finally:

    # Stop streaming
    pipeline.stop()
    cv2.waitKey(1)
    cv2.destroyAllWindows()
    cv2.waitKey(1)
```
Hope somebody could shed some light on how to load the json properly

closed 2021-03-22 by RealSenseSupport

sethuageye, 2021-09-01:
@RealSenseSupport, is there any official solution from your end?

maloel, 2021-09-01:
@sethuageye
AFAIK, with the last release of 2.49, using `serializable_device` should work.
Do you see otherwise?

sethuageye, 2021-09-01:
Can you point me to an example source code, which does that?

Sethu Subramaniam K.

maloel, 2021-09-01:
@sethuageye you can check out the code above from autimator. Or you can look at [one of the unit-tests](https://github.com/IntelRealSense/librealsense/blob/master/unit-tests/live/serializable-device/test-l500-json-load.py).

sethuageye, 2021-09-02:
It worked, appreciate the support. Thanks

Sethu Subramaniam K,
AIOT Developer,
AgEYE Technologies,
Bengaluru-560068.

estebandanielcaiza, 2021-09-24:
Hello
maybe @sethuageye have an code ?
I'm trying the solution of @autimator, but i have error.

CODE

import pyrealsense2 as rs
import numpy as np
import cv2
import json
import time

DS5_product_ids = ["0AD1", "0AD2", "0AD3", "0AD4", "0AD5", "0AF6", "0AFE", "0AFF", "0B00", "0B01", "0B03", "0B07", "0B3A", "0B5C"]

def find_device_that_supports_advanced_mode() :
    ctx = rs.context()  
    devices = ctx.query_devices()
    for dev in devices:       
        print("Found device:", dev.get_info(rs.camera_info.name))
        return dev 

# Configure
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
cfg = pipeline.start(config)

dev = find_device_that_supports_advanced_mode()
jsonObj = json.load(open("config1.json"))
json_string= str(jsonObj).replace("'", '\"')

ser_dev = rs.serializable_device(dev)  
with open(json_string, 'r') as file:
    json = file.read().strip()       
    ser_dev.load_json(json )

print("loaded json") 

try:
    while True:

        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()       
        color_frame = frames.get_color_frame() 
        color_image = np.asanyarray(color_frame.get_data())         

        cv2.imshow('RealSense', color_image)
        key = cv2.waitKey(1)
        # Press esc or 'q' to close the image window
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break

finally:

    # Stop streaming
    pipeline.stop()

-----------------------------------------------------------------------------------------------------------------

config.json  ---- from D435i

{
   "aux-param-autoexposure-setpoint":"1536",
   "aux-param-colorcorrection1":"0.298828",
   "aux-param-colorcorrection10":"-0",
   "aux-param-colorcorrection11":"-0",
   "aux-param-colorcorrection12":"-0",
   "aux-param-colorcorrection2":"0.293945",
   "aux-param-colorcorrection3":"0.293945",
   "aux-param-colorcorrection4":"0.114258",
   "aux-param-colorcorrection5":"-0",
   "aux-param-colorcorrection6":"-0",
   "aux-param-colorcorrection7":"-0",
   "aux-param-colorcorrection8":"-0",
   "aux-param-colorcorrection9":"-0",
   "aux-param-depthclampmax":"65536",
   "aux-param-depthclampmin":"0",
   "aux-param-disparityshift":"0",
   "controls-autoexposure-auto":"True",
   "controls-autoexposure-manual":"8500",
   "controls-color-autoexposure-auto":"True",
   "controls-color-autoexposure-manual":"156",
   "controls-color-backlight-compensation":"0",
   "controls-color-brightness":"0",
   "controls-color-contrast":"90",
   "controls-color-gain":"64",
   "controls-color-gamma":"300",
   "controls-color-hue":"0",
   "controls-color-power-line-frequency":"3",
   "controls-color-saturation":"64",
   "controls-color-sharpness":"100",
   "controls-color-white-balance-auto":"False",
   "controls-color-white-balance-manual":"5130",
   "controls-depth-gain":"16",
   "controls-laserpower":"150",
   "controls-laserstate":"on",
   "ignoreSAD":"0",
   "param-amplitude-factor":"0",
   "param-autoexposure-setpoint":"1536",
   "param-censusenablereg-udiameter":"9",
   "param-censusenablereg-vdiameter":"9",
   "param-censususize":"9",
   "param-censusvsize":"9",
   "param-depthclampmax":"65536",
   "param-depthclampmin":"0",
   "param-depthunits":"1000",
   "param-disableraucolor":"0",
   "param-disablesadcolor":"0",
   "param-disablesadnormalize":"0",
   "param-disablesloleftcolor":"0",
   "param-disableslorightcolor":"0",
   "param-disparitymode":"0",
   "param-disparityshift":"0",
   "param-lambdaad":"800",
   "param-lambdacensus":"26",
   "param-leftrightthreshold":"24",
   "param-maxscorethreshb":"2047",
   "param-medianthreshold":"500",
   "param-minscorethresha":"511",
   "param-neighborthresh":"7",
   "param-raumine":"1",
   "param-rauminn":"1",
   "param-rauminnssum":"3",
   "param-raumins":"1",
   "param-rauminw":"1",
   "param-rauminwesum":"3",
   "param-regioncolorthresholdb":"0.0499022",
   "param-regioncolorthresholdg":"0.0499022",
   "param-regioncolorthresholdr":"0.0499022",
   "param-regionshrinku":"3",
   "param-regionshrinkv":"1",
   "param-robbinsmonrodecrement":"10",
   "param-robbinsmonroincrement":"10",
   "param-rsmdiffthreshold":"4",
   "param-rsmrauslodiffthreshold":"1",
   "param-rsmremovethreshold":"0.375",
   "param-scanlineedgetaub":"72",
   "param-scanlineedgetaug":"72",
   "param-scanlineedgetaur":"72",
   "param-scanlinep1":"60",
   "param-scanlinep1onediscon":"105",
   "param-scanlinep1twodiscon":"70",
   "param-scanlinep2":"342",
   "param-scanlinep2onediscon":"190",
   "param-scanlinep2twodiscon":"130",
   "param-secondpeakdelta":"325",
   "param-texturecountthresh":"0",
   "param-texturedifferencethresh":"0",
   "param-usersm":"1",
   "param-zunits":"1000"
}

--------------------------------------------------------

estebandanielcaiza, 2021-09-24:
hello
@GreatScherzo can you help me with your json configuration file (LCustom.json) ?