realsenseai/librealsense #6903: D435激光功率问题

zouzoujia, 2020-07-23:
D435激光是红外光吧？光辐射功率多少？是否小于5mw/mm2
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RealSenseCustomerSupport, 2020-07-23:
----------------------------------------------

D435激光是红外光吧？

-是的  波長 850nm

光辐射功率多少？是否小于5mw/mm2

-D435 激光 是符合  激光安全 最高的 Class 1  等級  (裸眼)

Class 1, IEC 60825-1:2007 Edition 2, IEC 60825-1:2014
Edition 3

國內對應的就是 GB /T 7247.14 一類 激光產品

zouzoujia, 2020-07-24:
那光辐射功率是否小于5mw/mm2?对比了两个国标GB 、T 7247.14和我们这边需求的GB/T 3836.22-2017，没有查到类似数据。

zouzoujia, 2020-07-24:
另外还想问下相机可用的距离范围是多远？5m吗?

RealSenseCustomerSupport, 2020-07-27:
----------------------------------------------

-5m 還是會有深度影像但
基本上兩米內精準度可達2% RMS error 內, 隨著距離加長 error 變大 fill rate 變小
下面白皮書有到4 m 的 error 曲線
https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance

-辐射功率是小于5mw/mm2

RealSenseSupport, 2020-08-03:
Dear zouzoujia,
could we close this issue?

closed 2020-08-07 by RealSenseCustomerSupport