realsenseai/librealsense #6908: Issue with .get_timestamp() for frame synchronization

Erandurthil, 2020-07-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version |   Debian  | 
| Kernel Version (Linux Only)    |  5.3                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                        | 

### Issue Description
I want to synchronize poses from a T265 using interpolation of reported data in combination with the the timestamps of both devices. 
When I look at the metadata in the realsense viewer, the data is reported fine and even seems to be in Global Timestamp mode by default. 

When I try calling the .get_timestamp() function for a depth frame, a nullptr is returned. The function call for the pose frame works fine and seems to report correct data.

I've looked around and found the CMake flag ENFORCE_METADATA, but that seems to be for Windows systems, or I may have misread something.

dorodnic, 2020-07-23:
Hi @Erandurthil 
The error above corresponds to calling get_timestamp on an empty or uninitialized frame. Please check your application logic. You can check if the frame is valid like this - `if (frame) ts = frame.get_timestamp()`

Erandurthil, 2020-07-23:
Oh then that's my bad probably I'll check my code again, Thanks !

closed 2020-07-23 by Erandurthil