realsenseai/librealsense #6909: How can I program my code to save . Raw files instead of . PNG files

EggggggTart, 2020-07-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01,04,01,00 | 
| Operating System & Version |   Win 10                                     | 
| Platform                                 | PC  |
| SDK Version                            |  2.35.2.                           | 
| Language                            |  C++ }                          | 
| Segment			|  others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hello,everyone!
I want to get the angle of the camera while taking image.But I found that there was no function like stbi_write_png() to take the .raw file.
I need the .raw file to deal with depth information of image,and the angle is necessary to get the gradient of image.So using Realsense Wiewer or rs-convert.exe may cannot meet my needs.
What should I do?
EggggggTart.

EggggggTart, 2020-07-25:
I tried to read converter-raw.hpp ,and I used this code:

`std::ofstream fs(raw_file.str(), std::ios::binary | std::ios::trunc);
			if (fs) {
				fs.write(
					static_cast<const char *>(vf.get_data())
					, vf.get_stride_in_bytes() * vf.get_height());
				fs.flush();
			}`

But the .raw file I got in this way doesn't have the correct depth data.

Maybe I need more infomation about .raw files that saved by Realsense Viewer.

dorodnic, 2020-07-27:
Hi @EggggggTart 
I think the code is correct, what is vf? Note that you want to save the original depth frame, not its colorised representation.

EggggggTart, 2020-07-27:
> Hi @EggggggTart
> I think the code is correct, what is vf? Note that you want to save the original depth frame, not its colorised representation.

Hi!I'm glad that you can visit this issue.This code is comes from https://github.com/IntelRealSense/librealsense/blob/master/tools/convert/converters/converter-raw.hpp
Because of its function, I think I can find the answer from it. But I failed.

EggggggTart, 2020-07-28:
I found a function from https://github.com/IntelRealSense/librealsense/blob/master/common/model-views.cpp
`bool save_frame_raw_data(const std::string& filename, rs2::frame frame)
    {
        bool ret = false;
        auto image = frame.as<video_frame>();
        if (image)
        {
            std::ofstream outfile(filename.data(), std::ofstream::binary);
            outfile.write(static_cast<const char*>(image.get_data()), image.get_height()*image.get_stride_in_bytes());

            outfile.close();
            ret = true;
        }

        return ret;
    }`
Using this function can generate the .raw file I want correctly.The depth information in this .raw file is consistent with the depth information in the file obtained by using the RealSense Viewer.

Since I have solved this issue, I will close it.
I really appreciate dorodnic and other people who help others in forums.

closed 2020-07-28 by EggggggTart