realsenseai/librealsense #6910: RealSense Viewer - Display resolution

wjsprockel, 2020-07-23:
| | |
|---|---|
|**librealsense**|2.36.0 RELEASE|
|**OS**|Windows|

Having a display issue with RealSense Viewer on L515 on Windows 10.
Connecting the L515 for the very first time on my laptop was a real pleasure. Everything worked OK. Please see screencaps 'Laptop 1 & 2'. On Windows 10 Pro-64.
Closed application RealSense Viewer and the next time use I was having a display resolution problem. Please see screencap 'Laptop 3'. Like RealSense Viewer is taking over control of the display. I had no possibility to adjust or control this.
Next I connected the L515 on my workstation and ran RealSense viewer for the very first time. Here also, everything worked OK. Please see screencaps 'Workstation 1 & 2'. On Windows 10 Pro-64.
Then closed application RealSense Viewer. Started up again and I'm confronted with the same problem. Please see screencap 'Workstation 3'. 
RealSense Viewer is not displaying the complete screen. I cannot see everything and I cannot control or adjust anything.
I had the possibility to update the firmware, but this had no positive result. Please see screencap 'Update firmware'.
I tried everything in my possibility. Latest update Windows, remove/ install RealSense Viewer, etc.

Please: Any solution, any workaround? Any help will be greatly appreciated. 

PS. Very first time also on GitHub. I searched GitHub for known issues in this, with no result. 
I am a automotive technical trainer and I would like to use the L515 in my training and educational situation to demonstrate and let my students and trainees experience the laser distance measurement and point mapping. As the SSLaser is a very promising sensor for automotive ADAS.

Thank you all, Willy

[image]
[image]
[image]
[image]
[image]
[image]
[image]

dorodnic, 2020-07-23:
Hello @wjsprockel 
Welcome to this community, we will do our best to understand, reproduce and fix the issue. 
Between the first run and when the problem occurred did you make any changes to screen settings (connected to different monitor / docking-station)?
In between runs the RealSense Viewer is storing its settings in 'realsense-config.json' file. You can find it by entering '%APPDATA%' in file explorer. Deleting this file should reset the Viewer to its initial settings. Also, if you could attach its content before deleting it, it can shed some light on the source of the problem.

wjsprockel, 2020-07-23:
Hello Dorodnic, many thx for your reply.
I have just run RealSense Viewer. It is like forcing the display resolution to 640 x 480. So I am not getting the complete RealSense Viewer app window on display. And nothing I can do about that. When I switch back to windows my display resolution returns to my normal 3840 x 2160 resolution. But as soon as I switch to RealSense Viewer it goes back to 640 x 480, and only the upper left area of the display is in view.
Please find the config.json -file from just now, as a zip-file
[realsense-config.zip](https://github.com/IntelRealSense/librealsense/files/4966926/realsense-config.zip)
. And I also made a debug log, but it is way too big. Some 400 Mb. Maybe it is of some help.
Many thx. Willy

wjsprockel, 2020-07-23:
Hello Doronic,
"Deleting this file should reset the Viewer to its initial settings. ". Just to inform you.
I have deleted this -config.json-file. And run RealSense Viewer. Everything running OK again.
So I have some kind of a workaround for now. 
Many thx so far. Willy

dorodnic, 2020-07-23:
Thanks @wjsprockel 
I see the line: `"window.is_fullscreen": "1"`
Perhaps the Viewer was switched to full-screen mode by mistake (it can happen by pressing F8) and didn't identify the monitor resolution correctly. You should also be able to exit full screen by pressing F8 again, if the problem re-occurs.

wjsprockel, 2020-07-23:
Hello Dorodnic,
This works for now. I know now what to do. Using F8 to switch between full-screen mode and window.
Many thx, Have a nice evening & Stay safe. Willy

closed 2020-07-23 by wjsprockel