realsenseai/librealsense #6912: Segfault when importing pyrealsense2, after installing via pip3

matthewha123, 2020-07-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | None | 
| Firmware Version                   | None| 
| Operating System & Version |   Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  5.3.0-1023-aws                                        | 
| Platform                                 | AWS ec2 instance  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

I installed pyrealsense2 via pip3 on an ec2 instance. I am not using pyrealsense2 to read from cameras, but to read from .bag files recorded elsewhere.
A script that just runs:
```
import pyrealsense2
```
A seg fault occurs.
Here is the gdb backtrace:
```
Thread 1 "python3" received signal SIGSEGV, Segmentation fault.
__GI___pthread_mutex_lock (mutex=0x28) at ../nptl/pthread_mutex_lock.c:65
65	../nptl/pthread_mutex_lock.c: No such file or directory.
(gdb) backtrace
#0  __GI___pthread_mutex_lock (mutex=0x28) at ../nptl/pthread_mutex_lock.c:65
#1  0x00007ffff4f82c43 in libusb_get_device_list () from /opt/aivero/3dq/server/lib/libusb-1.0.so.0
#2  0x00007ffff599fe61 in librealsense::platform::usb_context::usb_context() ()
   from /home/matthewha123/.local/lib/python3.6/site-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#3  0x00007ffff58e3fc6 in librealsense::platform::usb_enumerator::query_devices_info() ()
   from /home/matthewha123/.local/lib/python3.6/site-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#4  0x00007ffff58e9382 in librealsense::platform::v4l_backend::query_usb_devices() const ()
   from /home/matthewha123/.local/lib/python3.6/site-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#5  0x00007ffff58fbae4 in librealsense::polling_device_watcher::polling_device_watcher(librealsense::platform::backend const*) ()
   from /home/matthewha123/.local/lib/python3.6/site-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#6  0x00007ffff58ecb9f in librealsense::platform::v4l_backend::create_device_watcher() const ()
   from /home/matthewha123/.local/lib/python3.6/site-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#7  0x00007ffff57e3166 in librealsense::context::context(librealsense::backend_type, char const*, char const*, rs2_recording_mode, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) ()
   from /home/matthewha123/.local/lib/python3.6/site-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#8  0x00007ffff5677c50 in rs2_create_context ()
   from /home/matthewha123/.local/lib/python3.6/site-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#9  0x00007ffff554d360 in ?? ()
   from /home/matthewha123/.local/lib/python3.6/site-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#10 0x00007ffff5590025 in ?? ()
   from /home/matthewha123/.local/lib/python3.6/site-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#11 0x00007ffff56524dd in ?? ()
   from /home/matthewha123/.local/lib/python3.6/site-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#12 0x00007ffff5522bd2 in PyInit_pyrealsense2 ()
   from /home/matthewha123/.local/lib/python3.6/site-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#13 0x00000000005fb2bf in _PyImport_LoadDynamicModuleWithSpec () at ../Python/importdl.c:159
#14 0x00000000005fb53d in _imp_create_dynamic_impl (module=<optimized out>, file=<optimized out>, 
    spec=<ModuleSpec(name='pyrealsense2.pyrealsense2', loader=<ExtensionFileLoader(name='pyrealsense2.pyrealsense2', path='/home/matthewha123/.local/lib/python3.6/site-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so') at remote 0x7ffff680c2e8>, origin='/home/matthewha123/.local/lib/python3.6/site-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so', loader_state=None, submodule_search_locations=None, _set_fileattr=True, _cached=None) at remote 0x7ffff680c358>)
    at ../Python/import.c:1994
#15 _imp_create_dynamic () at ../Python/clinic/import.c.h:289
#16 0x00000000005671ce in PyCFunction_Call () at ../Objects/methodobject.c:112
#17 0x0000000000511341 in do_call_core (kwdict={}, 
    callargs=(<ModuleSpec(name='pyrealsense2.pyrealsense2', loader=<ExtensionFileLoader(name='pyrealsense2.pyrealsense2', path='/home/matthewha123/.local/lib/python3.6/site-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so') at remote 0x7ffff680c2e8>, origin='/home/matthewha123/.local/lib/python3.---Type <return> to continue, or q <return> to quit---q
```

MartyG-RealSense, 2020-07-25:
Hi @matthewha123    Have you seen the past case in the link below about reading bags over an AWS EC2 connection, please?

https://github.com/IntelRealSense/librealsense/issues/4025

In that case, the RealSense user found that the problem they had was with corruption in the AWS versions of their files, whilst the locally stored versions of the files were fine.

matthewha123, 2020-07-27:
@MartyG-RealSense , while my overall goal is to read from a .bag file, the script that I am running only contains that import statement:
```
import pyrealsense2
```
Additionally, the error I am running into is a segmentation fault, which is not the same issue described in that other case (https://github.com/IntelRealSense/librealsense/issues/4025). That other case describes the library timing out when waiting for frames from a .bag.

MartyG-RealSense, 2020-07-28:
Apologies for the delay in responding further.  I was carefully researching your case.   There were a couple of other cases where RealSense users tried a minimal test script like the one above (except they used the full **import pyrealsense2 as rs**) and also got the segfault error.  No solution was found in those cases, unfortunately.

Regarding your problem with the bag, there was a case though where segfault was also occurring when extracting frames from a bag.  A RealSense user provided a script for a workaround that they developed to fix it.

https://github.com/IntelRealSense/librealsense/issues/885#issuecomment-349831693

@dorodnic the RealSense SDK manager provided some commentary further down from that linked-to comment.

https://github.com/IntelRealSense/librealsense/issues/885#issuecomment-352271685

MartyG-RealSense, 2020-08-08:
Hi @matthewha123   Do you require further support with this case please, or can it be closed?   Thanks!