realsenseai/librealsense #6913: Using brew to install glfw3 conflicts with FLAC on OSX Catalina 10.15.5

MichaelTiemann, 2020-07-23:
When ffmpeg is installed via brew on an OSX system, the FLAC subsystem is also installed.  The FLAC subsystem includes its own assert.h in /usr/local/include.

The realsense-viewer build targets search /usr/local/include so that they may find glfw3-related include files that are installed via brew.

Unfortunately, if ffmpeg is already installed on the system, then assert.h from FLAC is found before the assert.h presumed by imgui.h, and the build fails.

I think it's bad behavior on the part of FLAC to claim <assert.h> in the include hierarchy namespace, but there it is.

The work-around of uninstalling ffmpeg (and many other modules that depend upon FLAC) in order to clean up this conflict is annoying.  It would probably be better to move the very small glfw sub-system into the project build arena (rather than the /usr/local arena) so that /usr/local/include can be swapped out for /usr/include in the system include rules.

But I don't know that much about how brew and Xcode really should interoperate.  So an expert should weigh in.

MartyG-RealSense, 2020-07-24:
Hi @MichaelTiemann   I believe that the best way to handle your question will be to treat it as a Feature Request.  I will file a feature request report with Intel now and they can then direct the case to members of the RealSense team with the specialist knowledge to evaluate whether or not to proceed with implementing it.

MartyG-RealSense, 2021-11-16:
Closed due to decision not to proceed with implementing the requested feature.