realsenseai/librealsense #6915: Getting Intrinsic Values in Android Wrapper

joeyf123, 2020-07-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version | Android 10  |
| Kernel Version (Linux Only)    |  | 
| Platform                                 |  Android  |
| SDK Version                            |  2.36.0  | 
| Language                               |  Java | 
| Segment			        |  Robot | 

### Unable to retrieve Intrinsic information from the intrinsic class

Hello,

I am attempting to recreate the deproject method in the android wrapper. It appears that all the information necessary is available to me except the intrinsic information. While I am able to get the intrinsic class from the depth stream there is no way to extract that information as it is encapsulated in a way that makes it unreachable. It is inconsistent with its c++ equivalent because of this. What is in intention of this class and are getters able to be added to it so that we are able to make use of its data.

Thank you

ev-mp, 2020-07-23:
@joeyf123 , those Wrapper APIs have been recently added to the `development` branch  - https://github.com/IntelRealSense/librealsense/pull/6818
You can give it a try and provide feedback.

ev-mp, 2020-07-27:
The new wrapper functionalities are now part of the [v2.37.0 release](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#pre-release-2370)

closed 2020-07-27 by ev-mp