realsenseai/librealsense #6918: Motion Capture with L515

JeremyBYU, 2020-07-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 2.36 | 
| Operating System & Version |   Win 10S  | 
| Platform                                 | X86-64  |
| SDK Version                            |  2.36                          | 
| Language                            |  N/A                         | 
| Segment			|  Robot                         | 

### Qualisys Motion Capture Interferes with L515

Received the L515 yesterday and began some tests. I went out tonight into our outdoor facility that hosts over 30+ [Qualisys 7+ cameras](https://www.qualisys.com/hardware/5-6-7/) to see if if the L515 could operate in this environment. Unfortunately I believe the infrared light emitted from these motion tracking cameras is interfering with the L515. From what I can gather:

- Qualisys Cameras
    - The Qualisys camera uses short but quite strong infrared flashes to illuminate the markers. The flash is generated by LEDs on the front of the camera.
        - http://content.qualisys.com/2017/03/Getting-started.pdf, page 71
        - Can’t find the exact wavelength range they emit at, but see next bullet point for receiver wavelengths to get an idea
    - The Qualisys Receiver
        - Modulated light with a frequency of 33 kHz and is sensitive to light with wavelengths between 800 and 1100 nm.
        - Page 6
        - These flashes of infrared light from Qualisys are picked up the L515 receiver

- L515 Operates at 840-860 nm
- The banding of this light is picked up in the infrared sensor of the L515 and is clearly visible with modulated flashes and banding
- The generated depth map will have this banding for the grass area of the outdoor facility, but the white box was picked up just fine at 1.5 Meters. The center of the FOV is still the best quality.

I'm not sure that there is anything that can really be done about attenuating this type of interference. The only thing I could possibly think of is putting some sort of filter that blocks out anything above 900 nm over the L515 receiver. However I'm not sure if thats already been done, if it will even make a difference, or if this logic even makes any sense. I'm not faulting Intel here, just wondering if there are any other ideas of using the L515 with motion capture systems operating with near infrared. 

Also  was wondering if this type of interference would be expected on other LiDAR systems like Velodyne or Ouster? I've use Velodyne with Vicon and never had any issues. Maybe different wavelengths and different calibrated intensities for the beams. Or even just plain old different technology with Solid state LiDAR.  Thanks for any help or suggestions.

[image]

RealSenseSupport, 2020-07-24:
Hi @JeremyBYU 

Thanks for the testing information that you provided here.  We cannot comment on other LiDAR technologies but can only comment on our own L515 camera.

We don't have any specific testing information regarding filters over our camera sensors.  If you could, it would be a good experiment to test out various filters and see how they function.  As it stands, our camera operating in the 850nm range and any light source that emits in that range could cause interference.  

Not sure how much more I can help again as I cannot comment on other company's technologies.

JeremyBYU, 2020-07-24:
Okay sounds good. Appreciate your prompt response.

closed 2020-07-24 by JeremyBYU