realsenseai/librealsense #6919: DynamicCalibrator RGB image flickering / corrupted

nik518, 2020-07-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |   Win (10) / Linux (Ubuntu 18)| 
| Kernel Version (Linux Only)    | 5.3.0-40-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.33.10                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hi, I downloaded latest version of Dynamic Calibration Tool (2-11-0-0). I'm using D435i camera for calibration test's..
I noticed currupted images in RGB calibration phase (check attached image). The problem is more noticeable on linux than on windows..
Is this known issue? 
After a quick look into source available in Dynamic Calibration API, my guess would be that problem is in processing thread, where processing thread unlocks the working buffer while render thread is still using this buffer for rendering..

MartyG-RealSense, 2020-07-24:
Hi @nik518   Your kindly provided RGB image on Google Drive is set to  restricted access.  Can you set it to be shared with anyone please?  Thanks!

nik518, 2020-07-24:
> Hi @nik518   Your kindly provided RGB image on Google Drive is set to  restricted access.  Can you set it to be shared with anyone please?  Thanks!

link fixed..

MartyG-RealSense, 2020-07-24:
Thank you very much for the link fix.  I have posted the image below for ease of reference of others reading this case.

[image]

MartyG-RealSense, 2020-07-24:
If you are experiencing flickering, the first question I would ask is whether you have fluorescent lights such as ceiling strip-lights in the location where the camera is, please?

nik518, 2020-07-24:
> If you are experiencing flickering, the first question I would ask is whether you have fluorescent lights such as ceiling strip-lights in the location where the camera is, please?

Flickering is not the best word to describe my issue with DynamicCalibrator, we can focus on provided corrupted image..

MartyG-RealSense, 2020-07-24:
Okay, I see the problem in the image.  You are using the calibration print target, but the top of the RGB image is skewing off to the side so that the top of the target page is not aligned with the rest of the page.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/print-target-fixed-width.pdf

I have seen a few examples of this kind of corruption before, where sections of the image gets rearranged.  There isn't usually an obvious cause of it though.  Would it be possible to update your camera firmware to the newest **5.12.6** version please, to eliminate the possibility that it is a firmware related issue?

nik518, 2020-07-24:
> Okay, I see the problem in the image.  You are using the calibration print target, but the top of the RGB image is skewing off to the side so that the top of the target page is not aligned with the [...]

I can confirm that images received from callback (Rs400Device::StartCapture - m_colorSensor.start) are good, and images are still ok in processing thread, then somewhere in render thread images starts to become corrupted (probably becaouse of new image callbacks where new image data is written into buffer which is currently used by render thread - need to be confirmed..)

From provided image it is clearly visible that top-half of image contains old image data and bottom-half of image contains new image data..

MartyG-RealSense, 2020-07-24:
It is similar to a case I once had where the camera was moving down a hallway but the hallway position depicted on the top third of the RGB image was about a second ahead of the remaining lower two-thirds of the image that was lagging.  

Is the problem preventing dynamic calibration from completing successfully?

nik518, 2020-07-24:
> Is the problem preventing dynamic calibration from completing successfully?

No, calibration can be completed successfully, but sometimes it is very hard to navigate camera due heavy image curruption..

MartyG-RealSense, 2020-07-24:
Does the problem appear only during calibration or during normal use of the camera, please?

Also, what computer equipment are you using the camera with?  (PC, Mac, Raspberry Pi, Jetson, etc)

nik518, 2020-07-24:
> Does the problem appear only during calibration or during normal use of the camera, please?
During RGB calibration stage..

Required Info |  
-- | --
Camera Model | D400
Firmware Version | 05.12.05.00
Operating System & Version | Win (10) / Linux (Ubuntu 18)
Kernel Version (Linux Only) | 5.3.0-40-generic
Platform | PC
SDK Version | 2.33.10
Language | {C/C#/labview/nodejs/opencv/pcl/python/unity }
Segment | {Robot/Smartphone/VR/AR/others }

MartyG-RealSense, 2020-07-24:
Okay, so your computer is a full size desktop or laptop PC?

nik518, 2020-07-25:
> Okay, so your computer is a full size desktop or laptop PC?

Windows machine - YES, Linux machine - YES

MartyG-RealSense, 2020-07-25:
If you are finding calibration with the Dynamic Calibrator tool difficult due to this apparent lag in the RGB calibration, an alternative may be to use the newer **On-Chip Calibration** tool that is built into the RealSense Viewer program.

You can access this tool by going to the **More* option at the top of the Viewer's options side-panel and selecting **On-chip calibration*** from its drop-down menu.

A guide to using On-Chip calibration is in the link below:

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

closed 2020-08-08 by MartyG-RealSense