realsenseai/librealsense #6920: Recorder - Failed to convert source

jc211, 2020-07-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC |
| Language                            |  C++                          | 

### Issue Description
I am recording data without using the pipeline. I am using a device recorder, finding the profiles and sensors, opening and starting them. The result bag gives me a lot of error messages.

01:34:12 [Info] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\media\ros\ros_file_format.h:87 - Failed to convert source: rgb8 to matching rs2_format
01:34:12 [Info] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\media\ros\ros_file_format.h:65 - Failed to convert source: Raw Frame Size to matching enum rs2_frame_metadata_value

Any idea whats happening?

ev-mp, 2020-07-27:
@jc211 , those messages are marked as [info] and are primarily aimed at SDK developers.
While being rendered as inconvenient, the messages below [warning] do not affect the functionality and can be safely disregarded.

Are there any actual discrepancies in recorder that you observe?

RealSenseSupport, 2020-07-29:
Hi @jc211,

Has the comment provided by @ev-mp helped to find an answer to your question? Please let us know if you need any additional help with this matter.

Thank you!

jc211, 2020-07-29:
Yes thank you. I thought that those messages may be the reason I was getting frame drops. If I can continue in this thread, I'd like to ask about recording with the device directly as opposed to the pipeline. If I create a syncer object and attach it to the recorder's sensors, will the recorded data by synced? Does the pipeline method sync depth and color?

RealSenseSupport, 2020-07-29:
@jc211,

Since the original matter was resolved and you don't need further help with the question this thread was opened for - this  GitHub issue will be closed. For every additional question you have, please open a separate GitHub issue .

Thank you!

closed 2020-07-31 by jc211