realsenseai/librealsense #6922: ./scripts/patch-arch.sh patches fail on arch linux

LadaOndris, 2020-07-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR305 / D415 | 
| Firmware Version                   | ----------------
| Operating System & Version |   Arch Linux | 
| Kernel Version (Linux Only)    |  5.7.10-arch1-1                                        | 
| Platform                                 | PC  |
| SDK Version                            |  ----------------                         | 
| Language                            |  python                          | 
| Segment			|  other                      | 

### Issue Description
Hello, I'm trying to install librealsense according to https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md in conjuction with https://github.com/IntelRealSense/librealsense/issues/4271.

When I run ./scripts/patch-arch.sh I get the following output:
```
Extract the kernel
RealSense patch...
patching file drivers/media/usb/uvc/Makefile
Hunk #1 FAILED at 1.
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/usb/uvc/Makefile.rej
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 214 with fuzz 2 (offset 66 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 129 with fuzz 2 (offset 10 lines).
patching file include/uapi/linux/videodev2.h
Hunk #1 FAILED at 621.
1 out of 1 hunk FAILED -- saving rejects to file include/uapi/linux/videodev2.h.rej
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 2983 (offset 77 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 222 (offset 26 lines).
Prepare the build
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --syncconfig Kconfig
  HOSTCC  scripts/dtc/dtc.o
  HOSTCC  scripts/dtc/flattree.o
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/dtc/data.o
  HOSTCC  scripts/dtc/livetree.o
  HOSTCC  scripts/dtc/treesource.o
  HOSTCC  scripts/dtc/srcpos.o
  HOSTCC  scripts/dtc/checks.o
  HOSTCC  scripts/dtc/util.o
  LEX     scripts/dtc/dtc-lexer.lex.c
  YACC    scripts/dtc/dtc-parser.tab.[ch]
  HOSTCC  scripts/dtc/dtc-lexer.lex.o
  HOSTCC  scripts/dtc/dtc-parser.tab.o
  HOSTCC  scripts/dtc/yamltree.o
  HOSTLD  scripts/dtc/dtc
  HOSTCXX -fPIC scripts/gcc-plugins/structleak_plugin.o
  HOSTLLD -shared scripts/gcc-plugins/structleak_plugin.so
  HOSTCC  scripts/selinux/genheaders/genheaders
  HOSTCC  scripts/selinux/mdp/mdp
  HOSTCC  scripts/bin2c
  HOSTCC  scripts/kallsyms
  HOSTCC  scripts/recordmcount
[... 109 lines cut ...]
/home/lada/programs/librealsense/scripts/kernel/linux-5.7.10/drivers/media/usb/uvc//uvc_driver.c:265:11: error: ‘V4L2_PIX_FMT_INRI’ undeclared here (not in a function); did you mean ‘V4L2_PIX_FMT_INZI’?
  265 |   .fcc  = V4L2_PIX_FMT_INRI,
      |           ^~~~~~~~~~~~~~~~~
      |           V4L2_PIX_FMT_INZI
/home/lada/programs/librealsense/scripts/kernel/linux-5.7.10/drivers/media/usb/uvc//uvc_driver.c:270:11: error: ‘V4L2_PIX_FMT_INVI’ undeclared here (not in a function); did you mean ‘V4L2_PIX_FMT_INZI’?
  270 |   .fcc  = V4L2_PIX_FMT_INVI,
      |           ^~~~~~~~~~~~~~~~~
      |           V4L2_PIX_FMT_INZI
/home/lada/programs/librealsense/scripts/kernel/linux-5.7.10/drivers/media/usb/uvc//uvc_driver.c:275:11: error: ‘V4L2_PIX_FMT_RELI’ undeclared here (not in a function); did you mean ‘V4L2_PIX_FMT_INZI’?
  275 |   .fcc  = V4L2_PIX_FMT_RELI,
      |           ^~~~~~~~~~~~~~~~~
      |           V4L2_PIX_FMT_INZI
/home/lada/programs/librealsense/scripts/kernel/linux-5.7.10/drivers/media/usb/uvc//uvc_driver.c:280:11: error: ‘V4L2_PIX_FMT_Y8’ undeclared here (not in a function); did you mean ‘V4L2_PIX_FMT_Y8I’?
  280 |   .fcc  = V4L2_PIX_FMT_Y8,
      |           ^~~~~~~~~~~~~~~
      |           V4L2_PIX_FMT_Y8I
make[1]: *** [scripts/Makefile.build:267: /home/lada/programs/librealsense/scripts/kernel/linux-5.7.10/drivers/media/usb/uvc//uvc_driver.o] Error 1
make: *** [Makefile:1732: /home/lada/programs/librealsense/scripts/kernel/linux-5.7.10/drivers/media/usb/uvc/] Error 2
make: Leaving directory '/home/lada/programs/librealsense/scripts/kernel/linux-5.7.10'
cp: cannot stat '/home/lada/programs/librealsense/scripts/kernel/linux-5.7.10/drivers/media/usb/uvc/uvcvideo.ko': No such file or directory
Unloading existing uvcvideo driver...
modprobe: ERROR: could not insert 'uvcvideo': Unknown symbol in module, or unknown parameter (see dmesg)
Script has completed. Please consult the installation guide for further instruction.
```
It seems two patches failed to apply, therefore causing the compilation to fail.

MartyG-RealSense, 2020-07-25:
Hi @LadaOndris    If the patching process causes you problems, there is a **backend** method of installing librealsense that does not require patching.  It is also not dependent on Linux versions or kernel versions.  It requires an internet connection but can bet very useful in situations where patching or kernel issues are causing a librealsense build failure.

Once you have downloaded librealsense source code as a zip file from the SDK **Releases** page and extracted the zip's contents, you can install librealsense using the backend method described in the link below:

https://github.com/IntelRealSense/librealsense/issues/6710

LadaOndris, 2020-07-25:
@MartyG-RealSense Thanks for the answer. I successfully built it. How do I use it now? I'm trying to use pyrealsense2 package in python, but it says ``No device connected`` (source https://pypi.org/project/pyrealsense2/). I suppose it doesn't work just like that. What do you suggest?

MartyG-RealSense, 2020-07-25:
To use PyPi pip packages, you should have a computer with an x86 or x64 processor architecture (it will not work with devices with Arm processors).  Pip install also will not currently work with Python 3.8 as pyrealsense2 pip packages are only available up to Python 3.7 so far.

If you are using Python 3.8 then you can build the Pyrealsense2 wrapper from source code instead.

Alternatively, you can build librealsense using the backend method with the Python bindings included in the build.  You can do so by adding this flag to your CMake build statement.

-**DBUILD_PYTHON_BINDING=bool:true**

If you are building the bindings for Python 3 and you have both Python 2.7 and Python 3 installed on your computer, you may have to also add an additional flag called **-DPYTHON_EXECUTABLE**, otherwise pyrealsense2 may be built for Python 2.7 instead of 3.  Details about configuration of this are in the link below.

https://github.com/IntelRealSense/librealsense/issues/6449#issuecomment-658708484

Overall though, using Python 3.7 and pip install is likely to be the easiest install method.

LadaOndris, 2020-07-26:
@MartyG-RealSense It's not that I have a problem with installing the pyrealsense2. The problem is when I want to use it. It doesn't recognize any device and fails in ``pile.start()`` function.

MartyG-RealSense, 2020-07-26:
I see that you list both SR305 and D415 as your cameras at the top of this case.  Is it the SR305 that you have the connection problem with, or both of them please?

LadaOndris, 2020-07-26:
Both of them. I can see in dmesg the following output regarding usb:
```
[ 1111.520078] usb 2-2: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[ 1111.537877] usb 2-2: New USB device found, idVendor=8086, idProduct=0b48, bcdDevice=31.b3
[ 1111.537885] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1111.537890] usb 2-2: Product: Intel(R) RealSense(TM) Depth Camera SR305
[ 1111.537894] usb 2-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera SR305
[ 1111.537898] usb 2-2: SerialNumber: 636203005820
[ 1134.294452] usb 2-2: USB disconnect, device number 2
[ 1192.463264] usb 1-2: new high-speed USB device number 7 using xhci_hcd
[ 1192.604553] usb 1-2: New USB device found, idVendor=8086, idProduct=0ad3, bcdDevice=50.b1
[ 1192.604561] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1192.604565] usb 1-2: Product: Intel(R) RealSense(TM) Depth Camera 415 
[ 1192.604569] usb 1-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 415 
[ 1290.059659] usb 1-2: USB disconnect, device number 7
```

Before the process of patching the kernel, I saw in dmesg a line similar to this
``uvcvideo 2-6:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Camera was not initialized!``. But that is not the case anymore.

MartyG-RealSense, 2020-07-26:
Building librealsense with the backend method means that patching is not required.  So patch-related errors tend to disappear.

May I confirm that you are using the term **pipe.start()** in your program, as you earlier mention using **pile.start()** (which would be the incorrect term).

LadaOndris, 2020-07-26:
Sorry, that was a typo. I use the exactly same code as in https://pypi.org/project/pyrealsense2/.

MartyG-RealSense, 2020-07-26:
Is the camera detectable by pyrealsense2 if you do a complete shutdown of the computer (i.e not a reboot) and then reboot with the power button?

LadaOndris, 2020-07-26:
No luck. Still ``RuntimeError: No device connected``.

MartyG-RealSense, 2020-07-26:
It could be an udev rules problem specifically related to Arch Linux.  Please try the Linux command below to refresh the udev rules 

**udevadm control --reload-rules && udevadm trigger**

According to Dorodnic the RealSense SDK Manager, using pip install does not update the udev rules.  This may affect camera detection if the rules are out of date.

https://github.com/IntelRealSense/librealsense/issues/4011#issuecomment-493708239

LadaOndris, 2020-07-26:
Unfortunately, it still gives the same error. But I really appreciate the help.

MartyG-RealSense, 2020-07-26:
You are very welcome.  :)  Let's recap the details of the case.  Please let me know if I have anything incorrect.

-  You are using Arch Linux.
-  You installed librealsense using the backend method
-  You installed pyrealsense2 using pip install
- The camera can be detected with Arch Linux using dmesg but is not recognised by pyrealsense2

When you chose a pip **whl** file to install pyrealsense2, did you choose a filename that matches the number of your Python version?  **cp27** for Python 2.7,  **cp35** for Python 3.5 **cp36** for Python 3.6 and **cp37** for Python 3.7.

LadaOndris, 2020-07-26:
The recap is totally correct. 

To install pyrealsense2 using pip, I executed the command ``pip install pyrealsense2``. I didn't use any **whl** file.

LadaOndris, 2020-07-26:
I downloaded and installed ``pyrealsense2-2.36.0.2038-cp37-cp37m-manylinux1_x86_64.whl``, but the result stays the same.
I have Python 3.7.7.

MartyG-RealSense, 2020-07-26:
Oh yes, I believe that the whl file is chosen automatically from the list available depending on which Python version you are using (which is why it doesn't work with Python 3.8, because of the lack of a compatible package for 3.8).  Apologies for that.

MartyG-RealSense, 2020-07-26:
Aside from using the pip install method, did you have any luck with building the Python bindings alongside librealsense using CMake flags if you have tried that already?

https://github.com/IntelRealSense/librealsense/issues/6922#issuecomment-663891480

LadaOndris, 2020-07-26:
At the end of the installation using CMake, it outputs a warning:
```
CMake Warning:
  Manually-specified variables were not used by the project:

    BUILD_PYTHON_BINDING
```

MartyG-RealSense, 2020-07-26:
That is a very rare error.  Could you please use the command **clean cmake cache** and try the build command again?  I very much appreciate your patience.

LadaOndris, 2020-07-26:
I couldn't find a command as you said, so I simply deleted the whole folder, cloned the librealsense repo again and executed 
```cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_PYTHON_BINDING=bool:true``` 
in a build folder. 
Cmake outputs the same warning as before.

It is me who appreciates the help. :)

MartyG-RealSense, 2020-07-26:
It looks as though it is my own typo this time.  It should have been binding"s".  Sincere apologies!

**-DBUILD_PYTHON_BINDINGS=bool:true**

LadaOndris, 2020-07-26:
Now it successfully completed without warning. What should be the result of the command? What bindings does it create? I suppose I still need to use pyrealsense2 in python, am I right?

MartyG-RealSense, 2020-07-26:
It should have successfully built the pyrealsense2 wrapper.  

I believe that from this point on, point 4 of the instructions for building pyrealsense2 from source should apply.  I have pasted them below for your reading convenience.

*************
Update your PYTHONPATH environment variable to add the path to the pyrealsense library.

**export PYTHONPATH=$PYTHONPATH:/usr/local/lib**

LadaOndris, 2020-07-26:
So it finally works! :) To recapitulate what I just did to make it work for others in the future:

a) I followed the steps to **build pyrealsense2 wrapper from source**: https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python. I needed to call cmake with python bindings and then ``make -j4`` which created pyrealsense2 library files (.so files) in ``build/wrappers/python``.

b) Now, one has to **make sure Python knows about these libraries**. Because I use conda, there is, I believe, no reason to change PYTHONPATH, since conda does not use it. There are many ways to achieve this as described here https://stackoverflow.com/questions/37006114/anaconda-permanently-include-external-packages-like-in-pythonpath. I executed ``conda develop build/wrappers/python/`` which created .pth file in ``anaconda3/envs/ibt/lib/python3.7/site-packages`` folder.

From now on, I can ``import pyrealsense2 as rs`` and it works just fine. 

Thank you @MartyG-RealSense for all the help! :)

MartyG-RealSense, 2020-07-26:
Awesome news - thanks so much for the update and your detailed advice for others.   :)

closed 2020-08-08 by MartyG-RealSense