realsenseai/librealsense #6924: In the realsense viewer, My RGB feed keeps locking up and it says no frames received...now my depth feed is doing so as well

KEdwards79, 2020-07-26:
| | |
|---|---|
|**librealsense**|2.35.2 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|00000000f0210047|
|**Firmware Version**|01.04.01.00|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|00000003ac1e88b3|
|**Firmware Update Id**|00000003ac1e88b3|

Please provide a description of the problem

RealSenseSupport, 2020-07-27:
Hi @KEdwards79 

For full control/functionality of RealSense devices the SDK is the preferred method. You can however use RealSense Viewer to export a PointCloud file. Enable the RGB and Depth stream together then switch to the 3D view in the upper right corner. You should be able to see a textured point cloud of your scene. The export button above can save that to a PLY file which can then be read by supported software.

Thanks

RealSenseSupport, 2020-08-05:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

KEdwards79, 2020-08-11:
no, it turns out that the unit was overheating on the tripod.  I have since
made a fan-cooled case that works great!

closed 2020-09-14 by RealSenseSupport