realsenseai/librealsense #6927: Build fails with BUILD_EASYLOGGINGPP=OFF

alsora, 2020-07-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Operating System & Version |   Debian Buster  | 
| Platform                                 | Raspberry Pi/|
| SDK Version                            |  master                     | 

### Issue Description
Build fails when adding the CMake option `-D BUILD_EASYLOGGINGPP=OFF`

```
/root/librealsense/src/l500/ac-trigger.cpp:340:26: error: 'cout' is not a member of 'std'
                     std::cout << "-D- AC log is being written to: " << filename << std::endl;
                          ^~~~
/root/librealsense/src/l500/ac-trigger.cpp:340:26: note: 'std::cout' is defined in header '<iostream>'; did you forget to '#include <iostream>'?
/root/librealsense/src/l500/ac-trigger.cpp:11:1:
+#include <iostream>
 
/root/librealsense/src/l500/ac-trigger.cpp:340:26:
                     std::cout << "-D- AC log is being written to: " << filename << std::endl;
                          ^~~~
/root/librealsense/src/l500/ac-trigger.cpp: In member function 'virtual void librealsense::ivcam2::ac_logger::on_log(rs2_log_severity, const rs2_log_message&)':
/root/librealsense/src/l500/ac-trigger.cpp:350:13: error: 'log_message' was not declared in this scope
             log_message const & wrapper = (log_message const &)(msg);
             ^~~~~~~~~~~
/root/librealsense/src/l500/ac-trigger.cpp:350:13: note: suggested alternative: 'rs2_log_message'
             log_message const & wrapper = (log_message const &)(msg);
             ^~~~~~~~~~~
             rs2_log_message
/root/librealsense/src/l500/ac-trigger.cpp:351:32: error: 'wrapper' was not declared in this scope
             char const * raw = wrapper.el_msg.message().c_str();
                                ^~~~~~~
/root/librealsense/src/l500/ac-trigger.cpp:351:32: note: suggested alternative: 'toupper'
             char const * raw = wrapper.el_msg.message().c_str();
                                ^~~~~~~
                                toupper
/root/librealsense/src/l500/ac-trigger.cpp:359:22: error: 'cout' is not a member of 'std'
                 std::cout << text << std::endl;
                      ^~~~
/root/librealsense/src/l500/ac-trigger.cpp:359:22: note: 'std::cout' is defined in header '<iostream>'; did you forget to '#include <iostream>'?
make[2]: *** [CMakeFiles/realsense2.dir/build.make:557: CMakeFiles/realsense2.dir/src/l500/ac-trigger.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:74: CMakeFiles/realsense2.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

It looks like `log_message` from `log.h` is conditionaly compiled only when the logging flag is set to ON, but it's always compiled in `ac-trigger.cpp`

MartyG-RealSense, 2020-07-26:
Hi @alsora   The problem that you describe with **BUILD_EASYLOGGINGPP = OFF** is a known issue.  

https://github.com/IntelRealSense/librealsense/issues/6009

The problem was addressed via a "pull request" but apparently has not yet completed merging into the RealSense SDK.  

https://github.com/IntelRealSense/librealsense/pull/6304

alsora, 2020-07-27:
Thank you, I haven't seen the old issue.
We can close this.

closed 2020-07-27 by alsora