realsenseai/librealsense #6928: Can't build devleopment branch using Unity page instructions.

VisibleMeasurement, 2020-07-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.4.2.1| 
| Operating System & Version |   {Win 10)| 
| Kernel Version (Linux Only)    |  | 
| Platform                                 | PC  |
| SDK Version                            |  2.36 development branch}                          | 
| Language                            |  unity }                          | 
| Segment			|                            | 

### Issue Description
I am trying to run a build for the development branch to test if the L515 will work now.  I can get through the CMake step which creates the files in the build folder.  However, it then asks me to open the realsense.sln project, but there is only a file named "librealsense.sln".  If I open the file "librealsense.sln" it tells me it cannot reference "realsense.sln".

I then tried to change the target for the C# wrappers to .NET 3.5 and it tells me 3.5 will not run System.xaml.  If I run the build I get hundreds of errors and a failure.  

If I try to run the Unity examples without rebuilding the code, I have to use the old version of the unitypackage file and am assuming that's why I can't connect to the L515.

Any help?

Is it looking for the wrong files?

sldprt, 2020-07-26:
assembly name of intel.realsense.dll is wrong. the assembly name is currently "intel.realsensed" who realsensed this realsense in the past should fix it.

chrisrogers3d, 2020-08-03:
I tried to build 2.37 from github and had problems with atlcom.h and similar headers missing,.  I read some github issue here that said an atl installer had been added to your installer.  reinstalled everything including visual studio 2019 and still was not able to build.

I installed the binary 2.37(pre release) SDK installer and 2.37 unitypackage  and lidar works with realsense viewer but not unity..  unity 2019.3.13 Windows 10 L515 Fimrware 1.4.2.1  
even newer device D455 works with Unity on windows using the new 2.37 unitypackage.

RealSenseSupport, 2020-08-14:
Hi @VisibleMeasurement 

I see you were able to get things working based on your comments on the other github thread:  https://github.com/IntelRealSense/librealsense/issues/6859

There looks like others that were not able to get it working, if you can outline what you've done that would be great.

*If we don’t hear from you in 7 days, this issue will be closed.

VisibleMeasurement, 2020-08-17:
I did manage to get the L515 running in Unity, but I did not compile the code myself.  I used the newest unity package that was compiled by the RealSense team.  With that assest set loaded, I could open the default project provided and see the camera running in Unity.  Functionality was very similar to RealSense Viewer, so still have work to do before implementing anything more, like SLAM.

RealSenseSupport, 2020-08-17:
Thanks for the update @VisibleMeasurement

RealSenseSupport, 2020-09-11:
@VisibleMeasurement 

Is there anything else needed on this one?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-09-18 by RealSenseSupport