realsenseai/librealsense #6930: My pc can't seem to install the driver for the sensor

vertex798, 2020-07-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            | 2.0 | 
| Language                            |  { unity }                          | 
| Segment			|  {other }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hi guys!
I am going to use'RealSense L515'.
The driver was automatically installed by connecting the sensor to the first PC, but
I got the message'driver not available'.
And the dedicated viewer is also unable to find the device.
I am using Windows 10
Do you have anyone with the same symptoms as me?

dorodnic, 2020-07-27:
Hi @vertex798 
This can happen when the device is misidentified as USB2. Please check that the USB port and the cable are USB3 compatible.

vertex798, 2020-07-28:
p{margin-top:0;margin-bottom:0}
    

    Thanks!!I solved the problem with your help!!!
--------- 원본 메일 ---------

    
        보낸사람: Sergey Dorodnicov <notifications@github.com>
        받는사람: IntelRealSense/librealsense <librealsense@noreply.github.com>
        
            참조: vertex798 <himmel-aufstieg@kakao.com>, Mention <mention@noreply.github.com>
        
        날짜: 20.07.27 15:36 GMT +0900
        제목: Re: [IntelRealSense/librealsense] My pc can't seem to install the driver for the sensor (#6930)
        
        
    
    
        
Hi @vertex798
This can happen when the device is misidentified as USB2. Please check that the USB port and the cable are USB3 compatible.

closed 2020-07-28 by vertex798