realsenseai/librealsense #6935: How to get the intrinsics about realsense by C# ?

baifanxxx, 2020-07-27:
Hi,
I want to use C#  to develop realsense SR300. And I have a problem about how to get the intrinsics about realsense. I read the Intel® RealSense™ SDK 2016 R2 Documentation and found  PXCMScenePerception.GetInternalCameraIntrinsics. But, I do not use it successfully. Can you help me ? If you have a example code about it, it will be great.
Thanks.
BAI Fan

MartyG-RealSense, 2020-07-27:
Hi @baifanxxx   The '2016 R2' Windows SDK no longer receives technical support from Intel.  Support is provided for the current 'RealSense SDK 2.0' (librealsense), which is also compatible with the SR300.

https://github.com/IntelRealSense/librealsense

If you install the Windows version of RealSense SDK 2.0 then you can access a pre-made executable program called **rs-sensor-control**.  Because it is an executable, you do not need to compile it and so can run it immediately from the Windows command prompt window.

The command prompt window can be accessed on Windows by going to the text box at the bottom of your Windows screen (the one beside the button where you shut down the computer and typing **command** into the box to find the launch lnk for the command prompt.

Once in the command prompt window, you can find the sensor-control program by copy and pasting this command into the window:

**cd C:\Program Files (x86)\Intel RealSense SDK 2.0\tools**

This will take you to the RealSense SDK 2.0's "Tools" sample program directory.

After making certain that you have your RealSense camera plugged in, enter the command below to run rs-sensor-control to access the intrinsics and extrinsics by entering a series of option choices on your keyboard:

**rs-sensor-control**

A guide to using rs-sensor-control with SR300 is in the link below:

https://github.com/IntelRealSense/librealsense/issues/5134#issuecomment-546641014

Please let us know if you also need a way to access the intrinsics through C3 in RealSense SDK 2.0.

MartyG-RealSense, 2020-08-08:
Hi @baifanxxx   Do you require futher support with this case please, or can it be closed?   Thanks!

baifanxxx, 2020-08-10:
Thank you very muach. I finished it.

MartyG-RealSense, 2020-08-10:
Thanks so much for the update.

closed 2020-08-10 by MartyG-RealSense