realsenseai/librealsense #6936: Temperature measure using D435 camera relasense.

pp3993, 2020-07-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

pp3993, 2020-07-27:
Is it possible to know the tempratue with in 1 meter distance of person using this D435 camera? Please reply asap.

dorodnic, 2020-07-27:
Hi @pp3993 
No, these are not thermal cameras. The infrared band it can detect is much closer to the visible light.

pp3993, 2020-07-27:
is there any realsense series of camera that can do both work of detecting face and temprature as well?

MartyG-RealSense, 2020-07-27:
@pp3993  There was a case this month about synching a RealSense D435i camera with a FLIR camera.

https://github.com/IntelRealSense/librealsense/issues/6801

FLIR recently released a software product for their cameras that provides temperature sensing for COVID-19 screening.

https://www.theengineer.co.uk/supplier-network/product/flir-systems-launches-flir-screen-est-software-to-improve-skin-temperature-screening-for-covid-19/

MartyG-RealSense, 2020-08-08:
Hi @pp3993   Do you require further assistance with this case please, or can it be closed?   Thanks!

pp3993, 2020-08-12:
Thankyou @MartyG-RealSense . I am closing this now thanks for your help!!!!

closed 2020-08-12 by pp3993

MartyG-RealSense, 2020-08-12:
I'm pleased I could help - thanks for the update!