realsenseai/librealsense #6937: realsense-viewer setting option give error

tarekmuallim, 2020-07-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 | 
| Firmware Version                   | 05.12.06.00| 
| Operating System & Version |    Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    | 5.3.0-28-generic                                       | 
| Platform                                 | PC  |
| SDK Version                            | 2.36                           | 
| Language                            |  c++                         | 

### Issue Description
I have installed the packages using apt-get
the installation was fine
no error.

when I open the realsense-viewer I get this errors:

27/07 16:34:35,448 ERROR [140020130440960] (types.h:308) xioctl(VIDIOC_S_CTRL) failed Last Error: Invalid argument
 27/07 16:34:35,448 WARNING [140020130440960] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 27/07 16:34:35,479 ERROR [140020130440960] (types.h:308) xioctl(VIDIOC_S_CTRL) failed Last Error: Invalid argument
 27/07 16:34:35,479 WARNING [140020130440960] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance

and if I try to change any parameter value for example "Laser Power" I get an error:

Backend in rs2_set_option(options:0x7f58dc015ea0, option:Laser Power, value:300):
set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy

what is the problem and how to solve it.

[image]

MartyG-RealSense, 2020-07-27:
Hi @tarekmuallim   There was a similar past case in the link below.

https://github.com/IntelRealSense/librealsense/issues/5166

It is a very long discussion, so a good starting point may be the link below, which jumps to a point in the discussion that provides a list of instructions to try that will hopefully correct your problem.

https://github.com/IntelRealSense/librealsense/issues/5166#issuecomment-551066060

tarekmuallim, 2020-07-28:
Hi @MartyG-RealSense 

I tried this:

```
sudo apt-get purge librealsense2-dkms
sudo apt-get purge librealsense2-dbg
sudo apt-get purge librealsense2-dev
sudo apt-get purge librealsense2-utils
```

`sudo apt-get update --fix-missing && sudo apt-get install -f && sudo apt-get dist-upgrade -y 
`

`sudo apt-get install --reinstall linux-modules-extra-$(uname -r)
`

then I run this patch with no error
`./scripts/patch-realsense-ubuntu-lts.sh
`

```
sudo apt-get install librealsense2-dkms
sudo apt-get install librealsense2-utils
sudo apt-get install librealsense2-dev
sudo apt-get install librealsense2-dbg
```

and I still have the same error!

MartyG-RealSense, 2020-07-28:
I would expect to see VIDIOC errors as a result of a problem related to the Linux kernel.  There are a couple more cases in the links below that are similar to yours:

https://github.com/IntelRealSense/librealsense/issues/3426
https://github.com/IntelRealSense/librealsense/issues/5815

I believe that patching a DKMS package installation with **./scripts/patch-realsense-ubuntu-lts.sh** is not necessary, and instead is a command used doing a manual installation from source. 

One RealSense user got their installation working by purging old files and then doing a fresh install from the repositories.

https://github.com/IntelRealSense/librealsense/issues/3426#issuecomment-471760319
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

Alternatively, another installation method that you could consider is building librealsense from a source-code folder with a **backend** RSUSB installation after downloading a source-code zip file from the SDK releases page and extracting the librealsense folder from it.

https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279

tarekmuallim, 2020-08-05:
Thank you for your help @MartyG-RealSense 
building from source using `-DFORCE_RSUSB_BACKEND=true` solve the problem

closed 2020-08-05 by tarekmuallim

MartyG-RealSense, 2020-08-05:
I'm very pleased to hear that the method worked for you - thanks for the update!