realsenseai/librealsense #6938: Camera found by realsense-viewer but not via python and C++

PatrickNa, 2020-07-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9                                         | 
| Platform                                 | NVIDIA Jetson Xavier  |
| SDK Version                            |  2.36.0                          | 
| Language                            |  C++ and python3.6 }                          | 
| Segment			|  others                         | 

### Issue Description
I can not reach the camera via code. Neither with c++ nor with python. However it is available via realsense-viewer:
[image]

What am I missing here?

ev-mp, 2020-07-27:
@PatrickNa hello, 
When building from source c++/python modules make sure to set CMake with `-DFORCE_RSUSB_BACKEND=true` flag.
(or `-DFORCE_LIBUVC=true` for SDK prior to v2.30 as specified on [Jetson installation page](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md))). 
#6656

PatrickNa, 2020-07-28:
@ev-mp this worked. Thank you!
For everyone else, here are the steps I applied:

```bash
$ git clone https://github.com/IntelRealSense/librealsense.git
$ cd librealsense
$ mkdir build
$ cd build
$ cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DFORCE_LIBUVC=true -DBUILD_WITH_CUDA=true
$ make -j4
$ sudo make install
$ export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.6/pyrealsense2
```
Please note that I specified the python version when building. Also I exported the pythonpath to the spcific module, since it was not enough to just export to ```/usr/local/lib```.

closed 2020-07-28 by PatrickNa