realsenseai/librealsense #6940: Getting "terminate called after throwing an instance of 'rs2::backend_error'" error

milan-r-shah, 2020-07-27:
----------------------------------------------------------------------------------------------------

| Required Info                 |                                                    |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                    | 
| Firmware Version                   | 05.12.05.00          | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)   | 5.4.0-42-generic     | 
| Platform                                 | PC                           |
| SDK Version                          |  2.0 (v2.36.0)           | 
| Language                              |  C++                         | 
| Segment			        |  others                      | 

**Minimum FW Version** --> 05.12.05.00 

### Issue Description
I have been trying to execute **pcl-color** example by running the following command:
```bash
$ cd <librealsense>/build/wrappers/pcl/pcl-color
$ ./rs-pcl-color
```

But, every time, it give this error:
```bash
terminate called after throwing an instance of 'rs2::backend_error'
  what():  set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
Aborted (core dumped)
```

- On the other hand, the camera seems working properly in **realsense-viewer**.
- I tried building all the examples and pcl-wrapper-examples again but it didn't work except only for once.
- One more thing is, after getting this error for the first time, I noticed that if the camera gets tilted by more than 90&deg; then laptop display also gets rotated by right, left. or upside-down (Will create another issue for that).

MartyG-RealSense, 2020-07-27:
The first action I would recommend trying is changing your kernel from 5.4 Generic to 5.4 LTS if possible.

https://github.com/IntelRealSense/librealsense/issues/4818#issuecomment-529195542

milan-r-shah, 2020-07-28:
Thanks for the quick reply. 
However, in https://github.com/IntelRealSense/librealsense/issues/4818#issuecomment-529195542, what is **i86** in these sentences: "We only support official kernel versions used in i86 LTS versions of Ubuntu 16 / 18." and "For non-i86 platforms ..." ? 
Sorry, if this is silly question.

milan-r-shah, 2020-07-28:
The main reason for the confusion is that #4818 is mainly for Linux (Raspbian Buster) on Raspberry Pi4 while I'm using Ubuntu 18.04 LTS on a laptop (Dell XPS15 with i9 64-bit processor).

milan-r-shah, 2020-07-28:
> The first action I would recommend trying is changing your kernel from 5.4 Generic to 5.4 LTS if possible.
> [#4818 (comment)](https://github.com/IntelRealSense/librealsense/issues/4818#issuecomment-529195542)
> 
> The screen rotation question was replied to here:
> [#6941 (comment)](https://github.com/IntelRealSense/librealsense/issues/6941#issuecomment-664616359)

Changing the kernel from 5.4 Generic to 5.4 LTS seems little bit difficult thing so I ran the script mentioned in https://github.com/IntelRealSense/librealsense/issues/4818#issuecomment-529195542, but now, I'm getting the following error:

```bash
terminate called after throwing an instance of 'rs2::error'
  what():  No device connected
Aborted (core dumped)
```

MartyG-RealSense, 2020-07-28:
Apologies for any confusion.  The link was just to illustrate the point that it is preferable to use LTS kernels with librealsense.  A non LTS kernel may also work fine and some people may never experience a problem with it but there is a greater element of unpredictability.  Also, kernel 5.4 is not yet officially validated for use with librealsense (official support currently goes up to 5.3), adding a further element of unpredictability.

MartyG-RealSense, 2020-07-28:
I had also never heard of the term "i86" before, but research found that it is just a collective term for x86 and x64 processors).

http://www.edm2.com/index.php/I86

milan-r-shah, 2020-07-28:
> Apologies for any confusion. The link was just to illustrate the point that it is preferable to use LTS kernels with librealsense. A non LTS kernel may also work fine and some people may never [...]

Thanks for the suggestions. I reinstalled the entire Ubuntu 18.04 LTS and in the beginning, the kernel was **4.15.0-112-generic**. Then after reboot somehow it got changed to **5.4.0-42-generic** (maybe through previous grub recovery!). I tried removing that **5.4** one but couldn't. However, somehow I found a way to boot my Ubuntu from **5.3.0-28-generic** kernel. So, my current session of Ubuntu is running on **5.3**.

Last time, I had faced multiple problems while installing the SDK using pre-building packages, so, this time, I directly tried *backend* method i.e. [*installation.md*](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md). But, now, for `sudo apt-get install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev` command, I'm getting the following errors:
```bash
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version (1:2.17.1-1ubuntu0.7).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgtk-3-dev : Depends: libgtk-3-0 (= 3.22.30-1ubuntu1) but 3.22.30-1ubuntu4 is to be installed
                Depends: gir1.2-gtk-3.0 (= 3.22.30-1ubuntu1) but 3.22.30-1ubuntu4 is to be installed
                Depends: libgdk-pixbuf2.0-dev (>= 2.30.0) but it is not going to be installed
                Depends: libpango1.0-dev (>= 1.40.5) but it is not going to be installed
                Depends: libatk-bridge2.0-dev but it is not going to be installed
                Depends: libcairo2-dev (>= 1.14.0) but it is not going to be installed
                Depends: libx11-dev but it is not going to be installed
                Depends: libxext-dev but it is not going to be installed
                Depends: libxinerama-dev but it is not going to be installed
                Depends: libxi-dev but it is not going to be installed
                Depends: libxrandr-dev but it is not going to be installed
                Depends: libxcursor-dev but it is not going to be installed
                Depends: libxfixes-dev but it is not going to be installed
                Depends: libxcomposite-dev but it is not going to be installed
                Depends: libxdamage-dev but it is not going to be installed
                Depends: libegl1-mesa-dev but it is not going to be installed
                Depends: libxkbcommon-dev but it is not going to be installed
 pkg-config : Depends: dpkg-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```
What can/should I do now to install the SDK?

MartyG-RealSense, 2020-07-29:
Sincere apologies for the delay in responding further.  I have reviewed your case again from the beginning.  I note that your installation of librealsense by the **"backend"** method, whose instructions you provided a link to, is more commonly known as **building from source**.  Whilst doing so by this method has a number of advantages, it also has the disadvantage that you can end up with error messages if the kernel patching process does not go well.

There is another installation method that I refer to as **backend**, and it is performed over an internet connection.  Because of the way that it installs dependencies, it is not dependent on Linux versions or kernel versions and does not require patching.

The first step in this process is to download the source code of librealsense from the **Releases** page as a zip file.  The source code zip file can always be found in the "Assets" list at the bottom of the information listing for each SDK version on the Releases page.

https://github.com/IntelRealSense/librealsense/releases/

[image]

After downloading the zip file, its contents should be extracted so that you have a **librealsense** folder.At this point you can use the **RSUSB** installation method to install librealsense without dependence on Linux versions or kernel versions and without the need for patching.  This makes this installation method particularly suited to Arm devices such as Jetson.

The backend installation method requires an internet connection.  The steps are:

1.  Go to the librealsense root directory.  Create a folder within it called **build** and then go to this new folder using this instruction:

**mkdir build && cd build**

2.  Now that you are in the build directory, run the CMake build instruction below to install librealsense over the internet connection:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

If you need to install the PCL bindings at the same time as building librealsense, add this condition to the build statement above:

**- DBUILD_PCL_EXAMPLES=true**

MartyG-RealSense, 2020-08-08:
Hi @milan-r-shah   Do you require further support on this case please, or can it be closed?  Thanks!

milan-r-shah, 2020-08-11:
> Sincere apologies for the delay in responding further. I have reviewed your case again from the beginning. I note that your installation of librealsense by the **"backend"** method, whose [...]

Hi @MartyG-RealSense ,

Sincere apology for the late reply. 
Oh, I used to think that the **backend** method is as same as **building from the source** method. Because in other answers, I found it written as **backend method of installation from source**. Anyway, somehow, I solved that `unmet dependencies` error. 
But, I have one more confusion:
You had mentioned that kernel **5.4** is not supported yet but somehow, Ubuntu 18.04 LTS automatically updates the kernel to **5.4**. On the other hand, I couldn't find a proper way to downgrade it. So, currently, every time while logging in, I have to manually select the kernel **5.3** like this:

[image]

[image]

Apart from this, everything else seems to work fine. But, if I use **backend** method now, would I able to resolve this problem of manually choosing the kernel? Or there are chances that it would mess my current installation? (Because, last time, first I had tried installing the SDK using pre-built packages and it didn't work. So, then, I had tried installing it from the source. But, later one, every time, while opening `realsense-viewer`, I was getting `duplicate udev rules` warning/error)

MartyG-RealSense, 2020-08-12:
If you use kernel 5.4 then you will not necessarily encounter problems.   It just increases the risk a little that a problem might occur, since it is not officially validated for use with librealsense yet.  

The RSUSB backend method does build from a source-code folder but it installs dependencies over an internet connection, avoiding the need for patching the source-code build.  This can be very useful in situations such as installing librealsense on hardware that is experiencing conflicts with the Linux kernel.

There was an earlier installation method called **libuvc backend** that also uses an internet connection but has a different installation method, so some confusion is understandable.  It still works and is still useful in some circumstances.  

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

For PC though, I would recommend using the RSUSB backend method above.

https://github.com/IntelRealSense/librealsense/issues/6940#issuecomment-665713929

The link below discusses the advantages and disadvantages of using the backend method compared to kernel patching.  Scroll down the comment to the section of it headed  **What are the advantages and disadvantages of using libuvc vs patched kernel modules?**

https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604

RSUSB should be fine for use with a single camera and if your project is not a commercial one.  For a project that will become a commercial project that requires high stability and updatability though or one in which multiple cameras will be used, kernel patching is recommended.  There is also a higher risk of time-out errors when using RSUSB.

milan-r-shah, 2020-08-25:
Hi @MartyG-RealSense , sorry for the late reply.

You mentioned that
> For PC though, I would recommend using the RSUSB backend method described earlier in this discussion.
> 
> [#6940 (comment)](https://github.com/IntelRealSense/librealsense/issues/6940#issuecomment-665713929)
> 
But, in my case, I have already installed the SDK using **building from source** method. Last time, I was having certain problems due to **5.4** kernel. So, this time, I have to manually select kernel **5.3**. Therefore, now, if I try reinstalling the SDK using **RSUSB backend** method then would it work? or first I have to uninstall the existing one?

However, on the other hand, the project I'm working on is a commercial one so as you explained RSUSB method is not recommended, right? So, basically, right now, building from source is the only option for me, isn't it?

MartyG-RealSense, 2020-08-25:
The instructions for being librealsense in Linux recommend the DKMS distribution packages, with building from source available if there are certain circumstances in the project that require using that installation method instead.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

MartyG-RealSense, 2020-09-03:
Hi @milan-r-shah   Do you require further assistance with this case, please?   Thanks!

milan-r-shah, 2020-09-04:
Hi @MartyG-RealSense ,
Thanks a lot again for your concerns. In my case, installing the SDK using pre-built packages didn't work quite well last time. So then I had to install it from the source. But it had some issues with the Kernel version, in the beginning. 
As the project I'm working on is a commercial one, based on the above discussions, I assume that installing the SDK from the source would be an optimum option for me, at least for now. Right?

After giving you confirmation on this final query, please close this issue. 

Again, thank you so much @MartyG-RealSense for your time and for addressing all of my queries :)

MartyG-RealSense, 2020-09-05:
You are very welcome @milan-r-shah    :)

If it is a commercial project that requires stability and future upgradability then yes, using the **kernel patching** method would be the recommended action for the reasons described in the advantages / disadvantages section linked to earlier.

closed 2020-09-05 by MartyG-RealSense