realsenseai/librealsense #6941: Based on the camera orientation computer screen gets rotated (automatically)

milan-r-shah, 2020-07-27:
----------------------------------------------------------------------------------------------------

| Required Info                 |                                                    |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                    | 
| Firmware Version                   | 05.12.05.00          | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)   | 5.4.0-42-generic     | 
| Platform                                 | PC                           |
| SDK Version                          |  2.0 (v2.36.0)           | 
| Language                              |  C++                         | 
| Segment			        |  others                      | 

**Minimum FW Version** --> 05.12.05.00 

### Issue Description
Recently, after this #6940 error, I noticed that if **D435i** gets tilted by more than 90&deg; then laptop display also gets rotated by right, left. or upside-down as shown in the below pictures:

[image]

[image]

[image]

[image]

I don't know whether Is it a **realsense** issue or my laptop issue but it seems very strange. 

- I tried different USB ports but no luck.
- I tried restarting multiple times to confirm whether it's reproducible or not and it indeed occurred every single time.
- Even if I just connect the camera and don't launch anything, it affects my laptop screen orientation.

One more thing is, I'm still getting "Motion Module" error in *realsense-viewer* just like [this one](https://github.com/IntelRealSense/librealsense/issues/6784#issuecomment-662540099), as already mentioned in #6784 . At that time, I was told to ignore it. But just to confirm, the current issue can be the result of that error?

MartyG-RealSense, 2020-07-27:
I have seen a case of this screen rotation with the IMU before in https://github.com/IntelRealSense/librealsense/issues/6008

I recommended in that case that the RealSense user try turning off screen rotation in Ubuntu to see whether it resolved the problem, reasoning that it may be related to how a tablet changes screen orientation when rotated

https://askubuntu.com/questions/1035209/how-to-turn-off-screen-rotation-in-ubuntu-18-04-lts

Like yourself, that case involved motion module failure, suggesting a link between screen rotation and an IMU problem.

dorodnic, 2020-07-28:
From point of view of Laptop OS, IMU == screen rotator, similar to how it works in phones

MartyG-RealSense, 2020-08-08:
Hi @milan-r-shah   Do you require further assistance with this case please, or can it be closed?   Thanks!

milan-r-shah, 2020-08-11:
@MartyG-RealSense ,
Sincere apology for the late reply. As of now, turning off screen rotation in Ubuntu 18.04 solves the problem. So, closing this issue/case.

closed 2020-08-11 by milan-r-shah

MartyG-RealSense, 2020-08-11:
@milan-r-shah    Thanks very much!

pumpkin-ws, 2021-01-14:
I have the same problem on Ubuntu 20.04. I could turn off the screen rotation by locking the screen rotation, but realsense-reviewer still fires the following issue when trying to enable the motion module: 

Backend in rs2_open_multiple(sensor:0x7f2c84019080, profiles:0x7f2c851a57a8, count:2):
Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.5/0003:8086:0B5C.000A/HID-SENSOR-200076.3.auto/iio:device1/scan_elements/in_anglvel_x_en Last Error: Permission denied

I wonder if the imu on 455 is occupied by ubuntu automatically, and is there a way to have the system release control of imu?

MartyG-RealSense, 2021-01-14:
Hi @pumpkin-ws  I do not believe this is a problem that is specific to D455, as this error has occurred in the past with D435i (which also has an IMU).  It may be related to a problem with the **udev rules** that perform device handling, as described in the discussion in the link below.

https://github.com/IntelRealSense/librealsense/issues/5049