realsenseai/librealsense #6942: Connecting 8 L515 to one computer

jcova1996, 2020-07-27:
Hello,

I'm trying to create a system where I connect 8 L515 cameras to one computer. The 8 cameras are setup in 4 corners of a room, 4 being placed high, and 4 being placed low. The idea is to scan an object in the middle of the room with all 8 cameras at the same time. Having all 8 cameras stream depth and color images at the same time seems to cause some issues, as they drop frames or disconnect suddenly. They even sometimes disconnect when not in use. The cameras are connected with a USB - C to USB A cable, that connects to a USB hub with 4 USB A connections (UGreen brand), then connected to a USB A repeater cable (UGreen), then connected to Apple's USB - C to USB A / HDMI adapter, and finally to my MacBook running windows 10.

Is it possible to create a working setup from 8 L515 cameras streaming simultaneously to one computer?

 I'm using the latest SDK and coding in C#. This is my first time asking a question in GitHub, so please let me know if I need to include more details.

Thanks!

RealSenseSupport, 2020-08-07:
The WP about synchronizing multiple L515 cameras is now posted in the whitepaper section of our website. 
https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup

RealSenseSupport, 2020-08-18:
@jcova1996 

Hope the documentation will help you setup your configuration.

*If we don’t hear from you in 7 days, this issue will be closed.

jcova1996, 2020-08-20:
Yes, thank you!

RealSenseSupport, 2020-09-10:
@jcova1996 

Do you mean HW wise?  The setup from a HW standpoint can be seen in some of the other figures and images within the document.  From SW standpoint, the RealSense SDK was used.  Let me know if there is anything specific you were asking about.

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-09-17 by RealSenseSupport

jcova1996, 2020-09-17:
Sorry for the delayed response... Does this mean that the RealSense SDK can create meshes? Where in the SDK would I find this functionality?