realsenseai/librealsense #6943: Raspberry Pi 4: libuvc vs vl4c backend differences

dbolkensteyn, 2020-07-27:
This tutorial on how to use RealSense cameras on Raspberry Pi 4s mentions: https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras
> When building for the Raspberry Pi
-DFORCE_RSUSB_BACKEND=ON flag is also strongly recommended.

However this page on build options mentions: https://github.com/IntelRealSense/librealsense/wiki/Build-Configuration
> FORCE_LIBUVC
> libuvc can serve as a more robust alternative to the native backend, however, it has well known limitations **and is not recommended to be used in end-products** (please use FORCE_RSUSB_BACKEND cmake flag instead)

During my initials tests, I managed to get both `rs-enumerate-devices` and `rs-depth` work under the two modes.

Hence my questions:
1. What are the well-known limitations of libuvc, and why is it not recommended in end-products?
1. Why, specifically for the Raspberry Pi 4, is the libuvc mode recommended nevertheless?

Thank you for shedding some light on this!

MartyG-RealSense, 2020-07-27:
Hi @dbolkensteyn  For answers to your first question, please visit the link below and scroll down through the linked-to comment to the section headed **What are the advantages and disadvantages of using libuvc vs patched kernel modules?**

https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604

In regard to the second question, using libuvc on devices with Arm processors helps to avoid complications with kernel patching.

dbolkensteyn, 2020-08-02:
Thanks a lot @MartyG-RealSense 

My understanding after reading the comment you referenced is:
1. If no frame metadata beyond timestamps are needed, v4l2 on an unpatched kernel is the best option (as it has official support/maintainers, unlike libuvc)
1. If frame metadata are needed, then the ideal route would be to go for the patched kernel option which is officially supported and more reliable.
1. If that kernel patching is too demanding, or for debugging purposes, libuvc is an easier but unsupported route.

Is my understanding correct here?

MartyG-RealSense, 2020-08-03:
If you are seeking a highly stable production-level professional build then your understanding in each of your points is correct.   Though the patch-based build process is undoubtedly more complex than libuvc-backend / RSUSB backend over an internet connection, there are clear reasons why it is the recommended method in the documentation for building from source code even though an internet-based build is much simpler.

dbolkensteyn, 2020-08-03:
Thanks a lot @MartyG-RealSense , all clear

closed 2020-08-03 by dbolkensteyn