realsenseai/librealsense #6944: Framerate cant be controlled in Unity3D

masterchop, 2020-07-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265} | 
| Firmware Version                   | (0.2.0.951) | 
| Operating System & Version |   {Win (10) | 
| Platform                                 | PC |
| SDK Version                            |  { l2.35.2 }                          | 
| Language                            |  {UNITY}                          | 

### Issue Description

For some reason if i setup the FishEye Cameras to 10 FPS in unity i get the error:

ExternalException: rs2_pipeline_start_with_config(pipe:000000005BDEED70, config:000000005BDEEE10)
Rethrow as Exception: Couldn't resolve requests
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (IntPtr pNativeData)
Intel.RealSense.Pipeline.Start (Intel.RealSense.Config cfg)
RsDevice.OnEnable () (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:71)

I have to leave them to play at the OOTB FPS, is it possible to slow them down?
Is there any way to control the FPS in unity3d?

dorodnic, 2020-07-28:
Hi @masterchop 
The T265 device only supports 30 FPS for Fisheye cameras

masterchop, 2020-07-28:
what about the resolution, can we make it lower?

dorodnic, 2020-07-28:
848 X 800
Please review device data-sheet - https://dev.intelrealsense.com/docs/tracking-camera-t265-datasheet

closed 2020-08-04 by dorodnic