realsenseai/librealsense #6946: Connecting L515 where a D435 Just Was

lesreaper, 2020-07-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435/L515  | 
| Operating System & Version |   Jetson Nano Ubuntu 18.04  |                              | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  2.31, 2.37                         | 
| Language                            |  python                         |                  | 

### Issue Description
I have an entire setup already working with a D435 on a Jetson Nano and functioning perfectly. 

I tried plugging in an L515 (simple swap), reboot, and using the same cable, it didn't work. Couldn't find the device.

I upgraded to 2.37 because I thought there might be some library updates, but it still doesn't recognize the L515.

Is there a difference or addition one needs to go between the D435 and L515?

dorodnic, 2020-07-28:
If you are building the SDK from source please make sure to add `-DFORCE_LIBUVC=ON` flag to CMake and re-run setup of udev rules. With these steps done, L515 should work (with version 2.35.2 and up)
D435 can also work with`-DFORCE_LIBUVC=OFF` but it's not recommended.

lesreaper, 2020-07-29:
Tried, the `-DForce` was on, and used the L515 cable.

When I plug the L515 in, it's not recognized using `lsusb`.

I was hoping I could just unplug one and plug the other in, but it's not looking so easy.

RealSenseSupport, 2020-08-14:
Hi @lesreaper 

Were you able to get it working from "scratch", without doing just unplug plug swap?

*If we don’t hear from you in 7 days, this issue will be closed.