realsenseai/librealsense #6947: I am running this on amazon workspace windows 10 vm.  I just upgraded from 2.33 which was kindof working, it would at least open , although it would not open any files.

jamjahal, 2020-07-28:
| | |
|---|---|
|**librealsense**|2.37.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GLFW Driver Error: WGL: The driver does not appear to support OpenGL
Could not initialize offscreen context!

```

MartyG-RealSense, 2020-07-28:
For using a virtual machine with librealsense, Intel have separate advice in the first paragraph of the Windows and Linux documentation for compiling librealsense.

**Windows**
Use of a VM with librealsense is not recommended or supported due to issues with USB 3.

https://dev.intelrealsense.com/docs/compiling-librealsense-for-windows-guide

**Linux**
Intel again state non-support of VM installation but recommend that if you do try it then you use a  **VMware Workstation Player** virtual machine with librealsense to help ensure that it works with USB 3

https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide

dorodnic, 2020-07-28:
Hi @jamjahal 
The error message is accurate - The driver does not appear to support OpenGL.
Please make sure the graphics driver can run OpenGL to run graphic tools.

MartyG-RealSense, 2020-08-08:
Hi @jamjahal   Do you require further support with this case please, or can it be closed?   Thanks!