realsenseai/librealsense #6949: I can't connect to RealSense 515L from Unity.

vertex798, 2020-07-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { 515 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.0 | 
| Language                            |  unity }                          | 
| Segment			|  others     | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I can't connect to RealSense 515L from Unity.
In the current RealSense View, it has been confirmed that the sensor is operating normally.
However, after importing the Unity package file from the Unity project (Unity 2020.1.0f1) and executing the StartHere scene file, the following message appears.

[image]

ogoshen, 2020-07-30:
Looks like you're on a new version of Unity (2020)

Seems like the `ARBackgroundRenderer` has moved to a package:
https://docs.unity3d.com/Packages/com.unity.xr.arfoundation@3.1/api/UnityEngine.XR.ARFoundation.ARCameraBackground.html
The two aren't compatible though, the older is still available in Unity 2019, if that's an option for you..

vertex798, 2020-07-30:
@ogoshen  thank you
i used 
using UnityEngine.XR.ARFoundation

RealSenseSupport, 2020-08-14:
Were you able to get things working?

*If we don’t hear from you in 7 days, this issue will be closed.