realsenseai/librealsense #6953: How can map point from deep to rgb

solua100, 2020-07-28:
Hi, I face with a problem when try realsence.
How can I map a point(x,y) from deep data to a point(x,y) in RGB ?

For example: I detect a apple in rgb frame and want to calculate distance from cam to apple's center point.
I have center point (x,y) but I can't map it correctly to deep.  

I tried many example from realsence example, but they didn't fit my problem.
Can anyone give me idea to resolve it.
Thanks

MartyG-RealSense, 2020-07-28:
Hi @solua100   Which programming language are you using please (C++, C# or Python)?  Thanks!

solua100, 2020-07-28:
I use python.

MartyG-RealSense, 2020-07-28:
Intel have a Python tutorial called **distance_to_object** for aligning depth with color and displaying the distance to an object.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb

solua100, 2020-07-28:
Thanks MartyG-RealSense.
I will try it.

solua100, 2020-07-28:
It work !!!!!! 
Thanks MartyG-RealSense

closed 2020-07-28 by solua100

MartyG-RealSense, 2020-07-28:
Awesome news - thanks so much for the update!   :)