realsenseai/librealsense #6955: I need help to use it in unity!!

vertex798, 2020-07-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Win 10 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  Unity  | 
| Segment			|  Others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

What is cmake?
What do you say? in this part ....

[image]

Should I type it in cmd?
if so
''cmake' is an internal or external command, executable program, or
It is not a batch file.'
Only the message is displayed.
For reference, I have installed cmake.

Anyone would appreciate #6949 issues.

dorodnic, 2020-07-28:
Hi @vertex798 
The easiest way to get started with Unity and RealSense is by importing [UnityPackage](https://github.com/IntelRealSense/librealsense/releases/download/v2.37.0/Intel.RealSense.unitypackage).
There are some reports about issue with newly released Unity 2020.1 - we are looking into it. 
The package was tested with older version.

vertex798, 2020-07-28:
@dorodnic 
I tested it in Unity 2018.4.4f1, but I get the same symptoms.
Symptoms are reported in #6949.
#6949 Please provide feedback on the content.
**It is the same symptom in 2018.4.4f1 version.**

RealSenseSupport, 2020-08-14:
Were you able to get things working?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-09-11 by RealSenseSupport