realsenseai/librealsense #6956: [camera/camera_nodelet_manager-8] process has died - How to run SR300 with ros-kinetic-librealsense and librealsense2 on 4.15 kernel? (Turtlebot2i)

joana-ha, 2020-07-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300} | 
| Firmware Version                   | (3.17.0.0) | 
| Operating System & Version |   Linux (Ubuntu 16.04) | 
| Kernel Version (Linux Only)    |  4.15.0-112-generic                                   | 
| Platform                                 | NUC6CAYH |
| SDK Version                            |  librealsense2-dkms-1.3.12 with ros-kinetic-librealsense wrapper (uvcvideo-1.1.1-3-realsense)                         | 
| Language                            |  ROS                          | 
| Segment			|  Robot  (Turtlebot)                       | 

### Issue Description
1) installed librealsense2-dkms and librealsense2-utils (https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)
2) installed ros-kinetic-turtlebot* which installs ros-kinetic-librealsense and ros-kinetic-librealsense-camera as a dependency
3) While installing ros-kinetic-librealsense I was asked for patches for my kernel, skipped those
4) Ran turtlebot2i_bringup turtlebot2i_demo1.launch which, starts camera/camera_nodelet_manager and amongst others, includes a launchfile from realsense_camera package (sr300_nodelet_rgbd.launch), got the following warnings/errors:
```
[ WARN] [1595941674.520043333]: /camera_sr300/driver - Detected unvalidated firmware:
				- 627201001461's current camera firmware is 3.17.0.0, Validated camera firmware is 3.10.10.0
```
```
[ERROR] [1595941683.312106740]: /camera_sr300/driver - Error calling rs_wait_for_frames ( device:0x19b8100 ): 
Timeout waiting for frames. 
```
```
[camera/camera_nodelet_manager-8] process has died [pid 16751, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=camera_nodelet_manager __log:=/home/turtlebot/.ros/log/52f6d9ba-d0d3-11ea-9897-548d5a4379cf/camera-camera_nodelet_manager-8.log].
log file: /home/turtlebot/.ros/log/52f6d9ba-d0d3-11ea-9897-548d5a4379cf/camera-camera_nodelet_manager-8*.log
[ERROR] [1595941686.584203443]: Kobuki : malformed sub-payload detected. [107][170][6B AA 55 4D 01 0F DC 60 ]
```

I know the ros-kinetic-librealsense driver is outdated and probably not working for my kernel. I have a laptop on which I can run the driver successfully with a 4.4.0 kernel. Also I already managed to get the new ros wrapper realsense2_camera working on 4.15 kernel. However, in realsense2_camera there are not the same launchfiles available as in realsense_camera (e.g. sr300_nodelet_rgbd.launch). That's why I can't replace it. Is there a possibility to get ros-kinetic-librealsense and ros-kinetic-librealsense-camera work with my kernel?

If not, what can I do to fix this?

dorodnic, 2020-07-28:
Hi @joana-ha 
[rs_rgbd.launch](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_rgbd.launch) should be functionally similar and maintained for modern Kernel versions. 
realsense_camera is very outdated and there are no plans to backport 4.15 support into it. If using realsense2_camera is not possible, you should consider downgrading the Kernel version to 4.4

RealSenseSupport, 2020-07-31:
Hi @joana-ha,

Is there any specific reason to use kernel 4.15 in your system instead of kernel 4.4? If there is no such a reason, you should use kernel 4.4 in order to resolve the issue you're experiencing.
Please let us know if the comment provided by @dorodnic has helped to find an answer to your question
and if you need any further clarifications regarding this topic.

Thank you!

closed 2020-08-07 by RealSenseCustomerSupport