realsenseai/librealsense #6957: Problem with rs-convert to PLY

marteresagh, 2020-07-28:
| Required Info                         |                                                               |
|---------------------------------|-------------------------------------------|
| Camera Model                       | L515 | 
| Firmware Version                   | 01.04.01.02 | 
| Operating System & Version |   Windows 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.36.0                       | 

### Issue Description
Hi,
I recorded the streams as videos with RealSense Viewer and I obtained a `.bag` file. I tried to extract the frames with the rs-convert tool. The tool works for all formats except PLY: the process seems to run fine to the 100% PLY convert, but in the output folder there is no files saved. How can I resolve?

Thank you,
Maria Teresa

RealSenseSupport, 2020-08-13:
Hi @marteresagh 

Sorry for the delay, are you still having issues with producing PLY files from RealSense Viewer recordings?

I used on my Windows machine the following command line and was able to get generated PLY files:

C:\Program Files (x86)\Intel RealSense SDK 2.0\tools>rs-convert.exe -l test -i C:\temp\20200813_111127.bag

This was based on the 2.36.0 version of SDK using the rs-convert built executable.

*If we don’t hear from you in 7 days, this issue will be closed

marteresagh, 2020-08-17:
Hi,
yes, I still can't get the PLY files. I used the same tool, but in the folder I not find the files.

Thanks,
Maria Teresa

RealSenseSupport, 2020-08-17:
Hi @marteresagh 

Can you outline the steps in RealSense Viewer you are taking to record the bag files?  How about streams you have turned on or off prior to hitting the record button within the Viewer?  Can you provide screen captures of your Viewer setup?

Thanks

marteresagh, 2020-08-19:
I think I found the problem. I noticed that the .bag file, recording with only depth sensor activated, does not generate PLY files, I have to activate the rgb camera too.

Thank you

RealSenseSupport, 2020-08-19:
Sounds good.  Glad you were able to figure out the need for the rgb stream to be activated to get the PLY.

closed 2020-08-19 by RealSenseSupport