realsenseai/librealsense #6959: SR305 showing up as SR300 and RGB and depth stream locking up after seconds.

bucklevision, 2020-07-28:
| | |
|---|---|
|**librealsense**|2.29.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense SR300|
|**Serial Number**|635202001017|
|**Firmware Version**|3.27.3.0|
|**Camera Locked**|YES|
|**Product Line**|SR300|
|**Asic Serial Number**|635202001017|
|**Firmware Update Id**|635202001017|

Using the demo viewer v2.29.0 this device has started locking up after seconds. When switching to 3D mode it tells me the RGB stream has no frames received (from t0) and the depth stream runs for 3 seconds approx and then freezes.

MartyG-RealSense, 2020-07-28:
Hi @bucklevision   SDK 2.29.0 is very old, dating back to September 2019.  Unless there is a reason for using that specific version, I would recommend installing the full RealSense SDK version **2.36.0** using the link below.  This will place a launch icon on your Windows desktop for the 2.36.0 version of the RealSense Viewer program.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.36.0

If you are using Windows 10, you can find the installer program for 2.36.0 by going to the 'Assets' file list at the bottom of the information listing for version 2.36.0 in the above link and left-clicking on the file named **Intel.RealSense.SDK-WIN10-2.36.0.2034.exe** to launch a download of the 791 mb file in your browser.  When the download is complete, left-click on the downloaded file at the bottom of your browser window to run the installer.

[image]

bucklevision, 2020-07-29:
It is old, but had worked perfectly functionally previously so the only changes have been windows updates.

Having updated as you suggested last night, I now cannot execute the viewer at all (it opens momentarily, but before the realsense logo pops up, while it's still a bare white window it closes itself - with or without the camera attached, run with or without admin privs).

The depth quality tool loads, but has the original problem of just locking up after a second or so in 3D mode and showing no frames received in either window.

It's showing up as an SR305 now, but after a handful of frames it locks up until the camera is power cycled.
 
[image]
[image]

MartyG-RealSense, 2020-07-29:
If the RealSense Viewer and the Depth Quality Tool (which are structurally similar) are freezing up, this can indicate that the **OpenGL** graphics drivers on the computer may need updating.  This is because these programs depend on OpenGL to render the graphics.

You can usually update OpenGL simply by updating the graphics drivers on your computer.

bucklevision, 2020-07-29:
[image]
NVidia drivers are up to date, openGL not flagging any errors or compatibility issues (see attachment). Currently on latest graphics drivers, latest OpenGL, latest SDK and all Win10 updates done. Tried 2 different USB3 ports (both of which worked previously). Still the same error of no frames received.
Firmware version on the camera is 3.27.3.0.
Sometimes it freezes within the first few frames, sometimes it may be 100 frames in. No change of resolution or framerate seems to ease it, but it lasts longer on smaller frames at lower framerates on average.

MartyG-RealSense, 2020-07-29:
I ran tests with an SR300 and the Viewer of SDK 2.36.0 but could not replicate your problem.

Do you have the camera plugged into a USB port directly on the computer or is it plugged into a USB 3 hub, please?

bucklevision, 2020-07-29:
It's an SR305 (was originally showing up as a 300 for some reason, now shows up as a 305 with the latest SDK).

The port is a motherboard-mounted port, one of two USB3 ports on this board. The same ports that were used previously - I have tried both. The unit shows as present and seemingly correct in the device tree within win10 but I'm not sure where else to look. 

I've put the log level up to debug in the depth quality tool and taken a log, there's a lot of errors but without digging into the code yet I have no idea why.
[librealsense.log](https://github.com/IntelRealSense/librealsense/files/4996471/librealsense.log)

MartyG-RealSense, 2020-07-29:
My own camera is an original SR300 that identifies as an SR305 after I updated the firmware (the SR305 has an SR300 circuit board inside it).

Does it make any difference if you go to the **More** option at the top of the Viewer's options side-panel and select the **Hardware Reset** option from its menu to reset the camera before beginning streaming?

bucklevision, 2020-07-29:
I can't get into the viewer any more, it won't open - quick flash of a window and then closes itself. Back before I updated the SDK to solve this the hardware reset reset the camera, streamed for a few seconds and then stopped again.

MartyG-RealSense, 2020-07-29:
That is a weird behavior.  Does it also fail to launch if you go to the SDK folder at **C: > Program Files (x86) > Intel RealSense SDK 2.0 > Tools** and launch the Viewer from there by double-leftclicking on the **realsense-viewer** file?

bucklevision, 2020-08-03:
Sorry for the delay, had to move location.

Yes, fails the same way when run that way. Fails the same way run that way with "Run as administrator" and run through compatibility mode.

MartyG-RealSense, 2020-08-03:
There have been a few past cases where the Viewer would not launch if a **Documents** folder was not present in the user's My Documents folder.  Creating a folder called **Documents** if it was not there enabled the Viewer to launch.

Can you check your Windows user folder please to see if you have a **Documents** folder there?

[image]

bucklevision, 2020-08-03:
Present and correct as far as I can see.

[image]

The depth tool saved the log into that no problem, too.

MartyG-RealSense, 2020-08-03:
Let's go back to OpenGL testing.  Please return to the SDK's **Tools** folder and this time launch a program called **rs-capture**.  As this makes use of OpenGL like the Viewer does, if it works then it should help to eliminate OpenGL as the cause of the problem.

Below is an image of my SR300 (detected as an SR305) running **rs-capture**.

[image]

bucklevision, 2020-08-03:
This runs, but then the 3D stream fails after 1 second as on the depth tool, but the colour camera keeps on streaming. Colour FPS keeps updating ~1/sec, 3D stops updating at the last value.

[image]

MartyG-RealSense, 2020-08-03:
The error "Could not find camera in Windows device tree" makes me think that the SR305 may be disconnecting (the Device Tree is a system that keeps track of devices).  Rapid disconnection after connection, or not being detected at all, is a known problem with the SR30x cameras and can have numerous causes.

RealSense cameras of the SR300 generation are typically treated by the computer as ordinary USB video webcams if they have a problem, which may explain why RGB keeps functioning whilst depth (the RealSense element of the camera) does not.

bucklevision, 2020-08-03:
Interestingly it still shows up in the windows device manager as connected, and doesn't refresh as if dropping out, but is there a a way I can debug that more precisely?
[image]

MartyG-RealSense, 2020-08-03:
It is common that the SR3Ox will still be present in the Device Manager continuously but can deactivate as soon as it activates.  

The lights on the front of the camera can act as a debugging mechanism.  If the RGB stream only is active, there is a green light on all the time.  If a depth-sensing stream is active then there is also a red light on the front of the camera.

So if RGB was active but depth was not, I would expect to only see the green light remaining on.

bucklevision, 2020-08-03:
There are no lights on the SR305, I can just about make out the IR projector behind the body which is lit as long as the stream in rs-capture "should" be streaming - i.e. it fires up right away, but keeps going when the depth stream stops, until the software is closed by the red cross or by closing the command window.

MartyG-RealSense, 2020-08-03:
That suggests to me that the depth stream is active but has basically frozen.  So in the log line **RS frame rate printer**, the RGB FPS value keeps updating whilst the FPS of the depth stream is frozen at its last value (i.e not zero but not changing any more)?

bucklevision, 2020-08-03:
Yes, you've interpreted that perfectly.

MartyG-RealSense, 2020-08-03:
The next test, in the **Tools** folder again, would be to run the program **rs-hello-realsense**.  This is a text-only program that displays in text the currently detected distance and updates in real-time.  If this works fine then it will lend further weight to the problem being with OpenGL graphics, since this program does not use OpenGL

[image]

bucklevision, 2020-08-05:
This functions for approximately 1 second, rapidly updating the distance meansurement, and then freezes at a value, different value every time but accurate to the real distance.

MartyG-RealSense, 2020-08-05:
Thanks very much for testing.  Given that **rs-hello-realsense** does not use OpenGL, that swings the probabilities back towards a cause other than OpenGL.  

Since the camera is able to sense depth, if only for a brief time, it looks as though the camera drivers are okay.

Given that you are using Windows, it is worth checking the Windows camera privacy settings to see whether the camera is allowed to be used by applications.  The instructions for checking the permissions are in the link below.

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

bucklevision, 2020-08-05:
Hi, I hadn't considered that. 

I went to check the setting and it shows the camera as available, and accessible by all, see attached.
[image]

MartyG-RealSense, 2020-08-05:
Let's take the camera out of the equation for a moment.  You should be able to launch the RealSense Viewer without having a camera plugged in.   Is the Viewer able to complete launching if you remove the camera before running it, please?  If it still does not launch, this suggests that the problem is not with the camera hardware.

bucklevision, 2020-08-05:
ReaslSense Viewer behaves the same way as always, flashes up and closes itself with no error.
Depth Quality tool (without camera at start) sits with a blank white screen *sometimes* - once every 5 or 6 attempts it will get through to the splash screen and say it's waiting for a sensor, at which point if I plug the unit in it will progress. If the camera is already in, it goes straight through to the main screen every time but the camera locks up.

bucklevision, 2020-08-05:
At this stage I'm considering pulling the whole realsense SDK and drivers and re-installing on the assumption that something fell over during installation. Do you feel this is a valid route? Feels a bit "turn it off and on again".

MartyG-RealSense, 2020-08-05:
If you download the full RealSense SDK (791 mb) and run the installer, it should overwrite the previous installation.

You can also re-install the SR305 drivers by going to the Device Manager and right-clicking on each of the SR305 drivers and selecting **Uninstall** from the menu.  When all the SR305 drivers are removed, unplug the camera, wait a couple of seconds and plug the camera back in.  The drivers should automatically download and install over an internet connection.

Before doing that, you could try another test.  Download the standalone version of the RealSense Viewer called **Intel.RealSense.Viewer.exe**.  This is slightly different from the usual Viewer launch process, as you run the standalone Viewer from whatever folder location it is downloaded to instead of using the desktop launch icon.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.36.0

[image]

bucklevision, 2020-08-05:
Ooh interesting, the .exe version runs as per the depth tool - it starts up, runs both streams just fine and then the depth stream freezes.

MartyG-RealSense, 2020-08-05:
If you can at least get the standalone RealSense Viewer to finish launching and detect the camera then that is progress.

In that case, I would recommend going to the **Controls** section of the Viewer's options side-panel, finding the **Laser Power** slider and reducing its value down to around '1'.  Then start the depth stream and see if it still freezes.  

If it does not freeze, slowly increase the Laser Power value until it does freeze.

If you can stream at a low Laser Power value but not a higher value, this would suggest a problem with the camera not being supplied enough power for the demands that it is drawing.

In such circumstances, purchasing a mains electricity powered USB 3 hub and plugging the camera into it may provide additional power stability.

[image]

bucklevision, 2020-08-05:
I played with the power setting early on and it made no difference, with this thought in mind. I just tried it again and confirm it makes no difference still.  Hmm.

MartyG-RealSense, 2020-08-05:
If the camera worked fine until recently and then suddenly stopped working properly (what I call an "it worked until today" situation), it can indicate that Windows may have downloaded an update, or updated to a new version, and broke its RealSense camera support in the process.  It has happened numerous times in the SR300's history and there is nothing that Intel can do about it until Microsoft produce a Windows build that works with the camera again.

Do you know if Windows has done any obvious updates (such as when it tells you the % of an update that has installed) that coincided with the camera no longer working properly?

I am away for the next 7 hours now from the time of writing this but please do leave comments below and I will pick them up as soon as I return.

bucklevision, 2021-02-26:
Alas that's not really a solution. Problem: I bought a tyre for my car that is the right size but keeps going flat. Answer: Replace the car?

Honestly I expected a better product and better corporate support on a product carrying such a big name.