realsenseai/librealsense #6962: The realsense2.sln file is not being created.

vertex798, 2020-07-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Win 10 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  Unity 2020.1.0f.1  | 
| Segment			|  Others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I'm using Unity Wrapper according to your instructions (https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unity#step-1---build-dependencies-and-open-the-unity-project) I'm installing it.

But I ruined from the beginning.

I follow your manual
I followed "cmake .. -DBUILD_CSHARP_BINDINGS=ON -DBUILD_UNITY_BINDINGS=ON -DBUILD_SHARED_LIBS=ON -DDOTNET_VERSION_LIBRARY=3.5 -DCMAKE_GENERATOR_PLATFORM=x64"

The realsense2.sln file is not created.

I uploaded the video to Youtube and attach a link.
https://youtu.be/3ZGZM6VDDu0

Please solve the problem
(It made you work so complicated.)

The second ruined part is @dorodnic 's instructions.

He told me to use a version of Unity lower than 2020.1 in instructions from #6955 (https://github.com/IntelRealSense/librealsense/issues/6955#issuecomment-664890508).
So I tested on 2018.4.4f1.
The error disappeared, but the same symptom as #6949 appears.

I think this is the reason why CMake did not build.

Why did you make the sample project so complicated?

Can you update the fully running sample project?
(Even the published sample project has not been updated for 16 months.
You're selling new products, but
Your project doesn't fit the age at all)

* Moreover, the class ARBackgroundRenderer used in the sample is a deprecated class...

ev-mp, 2020-07-29:
@vertex798 , I will address the technical issue as highlighted in the log:
[image]

Cmake can't find **git**` required for building. It is either uninstalled or its path is not registered with **PATH** env. value.
You need to resolve it in order it to proceed with compilation.

vertex798, 2020-07-29:
@ev-mp Thank you for your answer.

By your instructions
The following issues were referenced (https://github.com/IntelRealSense/librealsense/issues/2719#issuecomment-474941250)

And the error no longer appears.

However, also the realsense2.sln file was not created.

What should I do?

Is it the process of creating the DLL to put in the Unity Asset/Plugins folder?
Can't I get the DLL file from you?

ev-mp, 2020-08-04:
@vertex798 , do you need further assistance with SLN?

RealSenseSupport, 2020-08-14:
Hi @vertex798 

Were you able to move forward with the suggestion from ev-mp?  Do you need anything else?

*If we don’t hear from you in 7 days, this issue will be closed.