realsenseai/librealsense #6963: Connect A RealSense Device

vertex798, 2020-07-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |    Win 10|
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC|
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            | unity 2017.4.13f1                          | 
| Segment			|  others                         | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Why do the following issues continue to occur?
(Unity 2017.4.13f1 : This is Unity for the project version you provided.)

[image]

dorodnic, 2020-07-29:
Hi @vertex798 
Did you try the package from [2.37 pre-release](https://github.com/IntelRealSense/librealsense/releases/tag/v2.37.0)?

vertex798, 2020-07-29:
@dorodnic 
is not??
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unity#unity-wrapper-for-realsense-sdk-20

vertex798, 2020-07-29:
It was solved with the help of @dorodnic.

closed 2020-07-29 by vertex798