realsenseai/librealsense #6964: Running pyrealsense2 on JetsonNano

fredy1221, 2020-07-29:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   |  | 
| Operating System & Version |  Linux Ubuntu 18 LTS | 
| Platform                                 | NVIDIA Jetson Nano |
| Realsense SDK Version        | legacy 2.36.0 | 
| Language                            | python | 
| Segment			| Industrial robot | 

### Issue Description

I'm using Intel Realsense D435i connected to a NVIDIA Jetson Nano.
The Jetson is flashed with Jetpack 4.3

First I installed RealSense viewer and it worked fine, I was able to stream from the camera in both depth and RGB.
Then I installed a pyrealsense2 python wrapper so I can start a pipeline to get a frame using a python script, now I can successfully import pyrealsense2 in my script but the device is no longer detected.

running: $ rs-enumerate-devices
returns: No device detected. Is it plugged in?
Even though the camera is plugged in

MartyG-RealSense, 2020-07-29:
Hi @fredy1221  The pyrealsense2 wrapper cannot be installed with the **pip install pyrealsense2** method on devices with Arm processors such as Jetson, because the PyPi pip packages are not compatible with Arm processors..  That typically means that the best option is to use **CMake** to build librealsense and the Python bindings at the same time.

The first step in this process is to download the source code of librealsense from the **Releases** page as a zip file.  The source code zip file can always be found in the "Assets" list at the bottom of the information listing for each SDK version on the Releases page.

https://github.com/IntelRealSense/librealsense/releases/

[image]

After downloading the zip file, its contents should be extracted so that you have a **librealsense** folder.At this point you can use the **RSUSB** installation method to install librealsense without dependence on Linux versions or kernel versions and without the need for patching.  This makes this installation method particularly suited to Arm devices such as Jetson.

The backend installation method requires an internet connection.  The steps are:

1.  Go to the librealsense root directory.  Create a folder within it called **build** and then go to this new folder using this instruction:

**mkdir build && cd build**

2.  Now that you are in the build directory, run the CMake build instruction below to install librealsense and the Python bindings over the internet connection:

make ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=...

The above statement is a basic one that should test whether the build is likely to succeed or not.  If it does succeed then you can try a more advanced build, which builds the example programs and includes optimizations such as building with CUDA support for faster alignment processing on devices such as Jetson that are equipped with an Nvidia graphics GPU.

cmake ../-DFORCE_RSUSB_BACKEND=ON  -DBUILD_PYTHON_BINDINGS:bool=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true

fredy1221, 2020-07-29:
Thank you for your help, it worked!

closed 2020-07-29 by fredy1221

MartyG-RealSense, 2020-07-29:
Excellent news - thank you very much for the update!   :)

ramgrandhi, 2020-08-01:
Hello @MartyG-RealSense - Thanks for your inputs so far! On a fresh install of JetPack 4.4 with Python3.7 as default (python3), I run into this issue while installing basic command you provided.

```
xxx@yyy:/librealsense-2.36.0/build$ **sudo cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3**
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.36.0
-- Setting Unix configurations
-- Info: Building with CUDA requires CMake v3.8+
-- CUDA_LIBRARIES: /usr/local/cuda/include /usr/local/cuda/lib64/libcudart_static.a;dl;/usr/lib/aarch64-linux-gnu/librt.so;/usr/local/cuda/lib64/libcusparse.so;/usr/lib/aarch64-linux-gnu/libcublas.so
-- using RS2_USE_LIBUVC_BACKEND
-- pybind11 v2.2.1
-- GLFW 3.3 not found; using internal version
-- Could NOT find Vulkan (missing: VULKAN_LIBRARY VULKAN_INCLUDE_DIR) 
-- Using X11 for window creation
CMake Error at third-party/glfw/CMakeLists.txt:235 (message):
  **The Xinerama headers were not found**

-- Configuring incomplete, errors occurred!
See also "/librealsense-2.36.0/build/CMakeFiles/CMakeOutput.log".
See also "/librealsense-2.36.0/build/CMakeFiles/CMakeError.log".
```

Highly appreciate your thoughts!

cheers.

MartyG-RealSense, 2020-08-01:
It's great that you quickly found a solution to your particular problem @ramgrandhi - thanks so much for sharing it!

ramgrandhi, 2020-08-01:
You're welcome @MartyG-RealSense! 

Though I can see pyrealsense2 packages installed, I still cannot import them from python3 REPL. Do you think I should explicitly create symlinks ? 

```
xxx@yyy:/$ **find . -name pyrealsense2 2>/dev/null**
./librealsense-2.36.0/build/wrappers/python/CMakeFiles/Export/lib/cmake/pyrealsense2
./librealsense-2.36.0/wrappers/python/pyrealsense2
```

cheers!

MartyG-RealSense, 2020-08-01:
@ramgrandhi  If you think that you need symbolic links (most people do not use them with Pyrealsense2), this may be a useful reference:

https://github.com/IntelRealSense/librealsense/issues/5275#issuecomment-565902666

Otherwise you could try this:

https://github.com/IntelRealSense/librealsense/issues/5275#issuecomment-555887940

Fred3D-tech, 2020-08-11:
it does not work for me on a Xavier Jetson with an updated config...
cmake -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=... -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true ..
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.36.0
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Setting Unix configurations
-- Info: Building with CUDA requires CMake v3.8+
-- The CUDA compiler identification is NVIDIA 10.2.89
-- Detecting CUDA compiler ABI info
-- Detecting CUDA compiler ABI info - done
-- Check for working CUDA compiler: /usr/local/cuda-10.2/bin/nvcc - skipped
-- Detecting CUDA compile features
-- Detecting CUDA compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda-10.2 (found version "10.2") 
-- CUDA_LIBRARIES: /usr/local/cuda-10.2/include /usr/local/cuda-10.2/lib64/libcudart_static.a;Threads::Threads;dl;/usr/lib/aarch64-linux-gnu/librt.so;/usr/local/cuda-10.2/lib64/libcusparse.so;/usr/lib/aarch64-linux-gnu/libcublas.so
-- Found usb: /usr/lib/aarch64-linux-gnu/libusb-1.0.so  
-- using RS2_USE_LIBUVC_BACKEND
-- Found Python: /usr/bin/python3.6 (found version "3.6.9") found components: Interpreter Development Development.Module Development.Embed 
-- Found PythonInterp: /home/fred/librealsense/build/...  
CMake Error at wrappers/python/third_party/pybind11/tools/FindPythonLibsNew.cmake:95 (message):
  Python config failure:

Call Stack (most recent call first):
  wrappers/python/third_party/pybind11/tools/pybind11Tools.cmake:16 (find_package)
  wrappers/python/third_party/pybind11/CMakeLists.txt:33 (include)

-- Configuring incomplete, errors occurred!
See also "/home/fred/librealsense/build/CMakeFiles/CMakeOutput.log".
See also "/home/fred/librealsense/build/CMakeFiles/CMakeError.log".

birdistheword96, 2020-10-01:
I am having the same issue, where I could originally see the stream from the camera by after installing pyrealsense2 wrapper it stopped working. 

I followed @MartyG-RealSense 's instructions (which were great btw) and got all the way through without any errors. However, when i run 'realsense-viewer' from command line it opens v2.31.0, rather than the version I have just complied with make/cmake which is v2.38.1.

I think having v2.31.0 is the problem as my L515 is not detected by it, so when i run any example python scripts (from the new folder) or the realsesense-viewer then the camera is not detected...

I'm sure many other people will find themselves n this page after having issues installing realsense, and have probably (like me) tried other tutorials that have installed other versions deep within the filesystem. Is there any easy way to remove the old version and swap it with the new one? and how to I link it to the 'realsense-viewer' command which is presumably calling the app from a link in /usr/bin?

MartyG-RealSense, 2020-10-01:
Hi @birdistheword96   Official support in the SDK for the L515  camera model began with version **2.35.2**.

If you built from source code with CMake then the link below provides guidance about where to find the built binary of the Viewer within the folder structure of the SDK version that you have just built.

https://github.com/IntelRealSense/librealsense/issues/1980#issuecomment-401599861

If you have 2.31.0 installed somewhere on your Jetson and you built it from source code, you could go to the **build** folder of that SDK and run this CMake command from that location:

**sudo make uninstall && make clean**

This should uninstall the SDK and clean up old files.

birdistheword96, 2020-10-01:
@MartyG-RealSense again thank you for the uninstall suggestion, this seems to have worked because I can run the realsense-viewer from the command line (although it reports to be 2.35.2, despite being built from 2.38.1) and I can now see the camera feed.

I tried one of the python examples and it said it couldn't import the module pyrealsense2, I think i read somewhere that I have to ad it to the python path or copy a .so file into the folder, is that correct?

MartyG-RealSense, 2020-10-01:
Hi @birdistheword96   Getting the Python wrapper working can depend on what term is used with **DPYTHON_EXECUTABLE** as this dictates how to find your Python installation.   If you have Python 3 installed and making **DPYTHON_EXECUTABLE** = '...' does not work, then a good starting point for testing is to use this term:

**DPYTHON_EXECUTABLE=/usr/bin/python3**

MartyG-RealSense, 2020-10-01:
Can you provide the error message that you get when you are unable to import, please?

birdistheword96, 2020-10-02:
Yeah sure, should have included it in the original message sorry:
```
user:~/librealsense-2.38.1/wrappers/python/examples$ python3 opencv_viewer_example.py

Traceback (most recent call last):
  File "opencv_viewer_example.py", line 8, in <module>
    import pyrealsense2 as rs
ModuleNotFoundError: No module named 'pyrealsense2'
```
I know that this error means it can't find the module to import, so I either need to add it to the python path or copy the library into the local folder where python file is. But which file is it that I need to copy specifically? or whoch directory shoudl i be adding to the python path and is there an easy way to do it?

Thanks for all your help so far @MartyG-RealSense!

MartyG-RealSense, 2020-10-02:
@birdistheword96   Reading back to your original comment about this problem in https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-702085783 I note that you are using the L515 model.  Could you therefore create a new issue about your current problem with **https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-702085783** and mention L515 in the case title please, as somebody else on the RealSense team other than myself is responsible or handling L515-related cases.  I do apologise.

birdistheword96, 2020-10-02:
@MartyG-RealSense No problem, will do.

Drkstr, 2020-10-11:
@MartyG-RealSense Thank you very much for the instructions above. 

Unfortunately, I am still having trouble importing pyrealsense2. Here are the steps I followed:

1.  Download the zip file from https://github.com/IntelRealSense/librealsense/releases/. I am using version 2.38.1

2. Extract the file and create a dir called build and cd into it.  
My directory structure for the build dir is `/home/rakshak/git_repo/librealsense-2.38.1/build `

3. Run the CMake command to test to see if the build will work. I used this command
`cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true `

4. Still in the build dir. Run `make -j4 `and then `sudo make install`
 
5. Add`export PYTHONPATH=$PYTHONPATH:/usr/local/lib` `to` the end of my .bashrc and source it. 

Now when I try to import it into python3, I get the following error. 

```
rakshak@RakshakNano:~$ python3
Python 3.6.9 (default, Jul 17 2020, 12:50:27) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2 as rs
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pyrealsense2'
```

I have also tried rebooting the Nano, still no luck. 

Let me know if you need any more information.

MartyG-RealSense, 2020-10-11:
Another RealSense user with a Jetson who had the **No module named 'pyrealsense2** error solved it by butting the two lines below **at the end** of their bash file as the final two lines:

**export PATH=$PATH:~/.local/bin
export PYTHONPATH=$PYTHONPATH:/usr/local/lib**

You can also check that your PYTHONPATH is set to the correct location by the bash file by using the command below after the bash script has been run:

**echo $PYTHONPATH**

Drkstr, 2020-10-11:
@MartyG-RealSense Thanks for the reply

I have added the following to the end of my .bashrc file and sourced the file

```
export PATH=$PATH:~/.local/bin
export PYTHONPATH=$PYTHONPATH:/usr/local/lib
```

`echo $PYTHONPATH` returns the following

```
rakshak@RakshakNano:~$ echo $PYTHONPATH
:/usr/local/lib
```
Still not able to import `pyrealsense2` 

Not sure what else to try at this point 😿

MartyG-RealSense, 2020-10-11:
You could next try targeting Python 3.6 specifically with **DPYTHON_EXECUTABLE=/usr/bin/python3.6**

If that does not work, there was a Jetson case with this error a couple of months ago that was eventually solved by updating CMake.  The RealSense user shared the process they used for updating CMake on their Jetson in the link below:

https://github.com/IntelRealSense/librealsense/issues/6980#issuecomment-666858977

Drkstr, 2020-10-13:
@MartyG-RealSense That did the trick. 

For future reference, here is a complete list of steps I took to get the IntelRealsense Camara D455 to work with python3 on the Jetson Nano. 

1. Update CMake using [this method](https://github.com/IntelRealSense/librealsense/issues/6980#issuecomment-666858977)

2. Download the zip file from https://github.com/IntelRealSense/librealsense/releases/. I am using version 2.38.1

3. Extract the file, cd into the extracted file .

4. Create a dir called build and cd into it.  

5. Run the CMake command to test to see if the build will work.
 `cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true - DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true`

6.  Still in the build dir. Run make -j4 and then sudo make install.

7. Add these to the end of your .bashrc file
```
export PATH=$PATH:~/.local/bin
export PYTHONPATH=$PYTHONPATH:/usr/local/lib
```

8. source your .bashrc file

9. Deal with package restructuring between python2 and python3 using [this method](https://github.com/IntelRealSense/meta-intel-realsense/issues/20#issuecomment-687180484)
You should now be able to import` pyrealsense2` using python3 

@MartyG-RealSense thank you once again for your help 💯

MartyG-RealSense, 2020-10-13:
Hi @Drkstr  Thanks so much for sharing your detailed process with the community - great to hear that you were successful!

Drkstr, 2020-10-21:
Hey, @MartyG-RealSense am now trying to get this to work on the NX. 
I've followed all the steps from my post earlier on this thread and I have tried everything from this [thread](https://github.com/IntelRealSense/librealsense/issues/6980). 

But am still getting a "ModuleNotFoundError: No module named 'pyrealsense2'"  

Do you have any other suggestions for me?

MartyG-RealSense, 2020-10-21:
@Drkstr   It looks as though you found the solution in the other thread with the help of GOBish

https://github.com/IntelRealSense/librealsense/issues/6980#issuecomment-713321354

Drkstr, 2020-10-22:
Yup, got it sorted :D

MartyG-RealSense, 2020-10-22:
Great @Drkstr   Thanks!

FrankCreen, 2020-11-16:
> @MartyG-RealSense That did the trick.
> 
> For future reference, here is a complete list of steps I took to get the IntelRealsense Camara D455 to work with python3 on the Jetson Nano.
> 
> 1. Update CMake using [this method](https://github.com/IntelRealSense/librealsense/issues/6980#issuecomment-666858977)
>    (You need to have curl [installed](https://stackoverflow.com/questions/56941778/cmake-use-system-curl-is-on-but-a-curl-is-not-found) for this to work. )
> 2. Download the zip file from https://github.com/IntelRealSense/librealsense/releases/. I am using version 2.38.1
> 3. Extract the file, cd into the extracted file .
> 4. Create a dir called build and cd into it.
> 5. Run the CMake command to test to see if the build will work.
>    `cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true - DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true`
> 
> If you have issues with the above step, check out [this](https://github.com/jetsonhacks/buildLibrealsense2TX/issues/13), [this](https://github.com/IntelRealSense/librealsense/issues/6449#issuecomment-650793097) and [this](https://askubuntu.com/questions/620601/cant-install-libsdl2-dev-or-cmake-glfw-libxinerama-dev-problem)
> 
> 1. Still in the build dir. Run` make -j4` and then `sudo make install`.
> 2. Add these to the end of your .bashrc file
> 
> ```
> export PATH=$PATH:~/.local/bin
> export PYTHONPATH=$PYTHONPATH:/usr/local/lib
> export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.6/pyrealsense2
> ```
> 
> 1. source your .bashrc file
> 2. Deal with package restructuring between python2 and python3 using [this method](https://github.com/IntelRealSense/meta-intel-realsense/issues/20#issuecomment-687180484)
> 
> You should now be able to import` pyrealsense2` using python3
> 
> @MartyG-RealSense thank you once again for your help 💯
> 
> Update: I can confirm that this works for the Jetson Xavier NX as well.

Hi~ @Drkstr  My occasion is very similar to you. My camera is D455 and the compute device is Jetson Xaiver AGX. My task is using the C++ interface of librealsense-2.38.1 to caputre video frame and do some image process. My problem is that I can use the realsense-viewer  get a 90FPS video frame sequence at 480*270 resolution,but I only got a 54FPS using the C++ interface. I tested the same code on two xiver AGX with the same results. I don't know what should I do next step? Can you give me some advices? Thank you!

```
// include the librealsense C++ header file
#include <librealsense2/rs.hpp>

#include <opencv2/opencv.hpp>

#include <chrono>

using namespace std;
using namespace cv;

int main()
try
{
    //Contruct a pipeline which abstracts the device
    rs2::pipeline pipe;

    //Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;
    //Add desired streams to configuration
    cfg.enable_stream(RS2_STREAM_COLOR, 480, 270, RS2_FORMAT_BGR8, 90);

    //Instruct pipeline to start streaming with the requested configuration
    pipe.start(cfg);

//Sensor Settings
    //Depth Camera Settigs
    auto sensor11 = pipe.get_active_profile().get_device().query_sensors()[0];
    sensor11.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);

    //RGB Camera Settigs
    auto sensor2 = pipe.get_active_profile().get_device().query_sensors()[1];
    sensor2.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
    sensor2.set_option(RS2_OPTION_AUTO_EXPOSURE_PRIORITY, 0);
    sensor2.set_option(RS2_OPTION_ENABLE_AUTO_WHITE_BALANCE, 1);
    //Controls Settings
    sensor2.set_option(RS2_OPTION_BRIGHTNESS, 0.0);
    sensor2.set_option(RS2_OPTION_CONTRAST, 50.0);
    sensor2.set_option(RS2_OPTION_EXPOSURE, 160.0);
    // sensor2.set_option(RS2_OPTION_GAIN,64.0);
    sensor2.set_option(RS2_OPTION_GAMMA, 300.0);
    sensor2.set_option(RS2_OPTION_HUE, 0.0);
    sensor2.set_option(RS2_OPTION_SATURATION, 64.0);
    sensor2.set_option(RS2_OPTION_SHARPNESS, 50.0);
    sensor2.set_option(RS2_OPTION_WHITE_BALANCE, 4600.0);

    sensor2.set_option(RS2_OPTION_POWER_LINE_FREQUENCY, 0);
    sensor2.set_option(RS2_OPTION_GLOBAL_TIME_ENABLED, 1);

    // Display in a GUI
    const auto window_name = "Display Video";
    namedWindow(window_name, WINDOW_AUTOSIZE);

    //用于计时
    //show time
    auto start = chrono::system_clock::now();
    int frame = 0;

    while (waitKey(1) < 0 && getWindowProperty(window_name, WND_PROP_AUTOSIZE) >= 0)
    {
        rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera

        //获取彩色图像帧
        rs2::frame color = data.get_color_frame();

        //保存彩色图像
        Mat image(Size(480, 270), CV_8UC3, (void *)color.get_data(), Mat::AUTO_STEP);

        // Update the window with new data
        imshow(window_name, image);

        frame++;
        //one loop end
        auto end = chrono::system_clock::now();
        auto duration = chrono::duration_cast<chrono::microseconds>(end - start);
        double timecost = double(duration.count()) * chrono::microseconds::period::num / chrono::microseconds::period::den;

        cout << frame / timecost << " FPS" << endl;
    }

    return 0;
}
catch (Exception ex)
{
    cout << "Error!!!" << endl;
    cerr << ex.msg << endl;
}
```

Drkstr, 2020-11-16:
@FrankCreen sorry man, I have no experience with c++.

Schloern93, 2020-12-01:
Hello,
When executing the command:

cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true - DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true

I got the following error:

CMake Error: The source directory "/home/jetson/fehnle/locale_work/install_librealsense/librealsense-2.38.1/build/DPYTHON_EXECUTABLE=/usr/bin/python3.6" does not exist.

What am I doing wrong?
Many thanks in advance

barTRIP, 2021-03-01:
I can confirm these instructions also work for the Raspberry Pi 4 B (8Gb) running on Ubuntu Server 20.04.2. 

> @MartyG-RealSense That did the trick.
> For future reference, here is a complete list of steps I took to get the IntelRealsense Camara D455 to work with python3 on the Jetson Nano.
>
> 1. Update CMake [...]

MartyG-RealSense, 2021-03-01:
Thanks so much @barTRIP for adding your compatibility information!

aloev, 2021-04-08:
In my case I'm using Jetson nano with an intelrealsense 435. I'm running ubuntu 18.04 with python 3.6 so I followed @Drkstr steps on how to install pyrealsense properly and it did but when I run my scripts I get the following

[image]

Also when I try to run it on the terminal I get

[image]

To my surprise the terminal window that I used to install pyrealsense2 is the only one that recognizes the command

[image]

I'm still new into all this so If anyone knows what is going on and can point out to me what I'm missing I would highly appreciate it.

samwesley, 2021-06-10:
> @MartyG-RealSense That did the trick.
>
> For future reference, here is a complete list of steps I took to get the IntelRealsense Camara D455 to work with python3 on the Jetson Nano.
>
> 1. Update CMake [...]

On **Step 5**, I was running into the following error: "CMake Error: The source directory "/home/pi/librealsense-2.47.0/build/DPYTHON_EXECUTABLE=/usr/bin/python3.6" does not exist."

It turns out there was a space between "-" and "DPYTHON_EXECUTABLE=/usr/bin/python3.6" below:

cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true - DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true

This was fixed by removing the space :)
`cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true`

MartyG-RealSense, 2021-06-10:
Thanks very much @samwesley for adding your experience to this installation guide. :)

zebrassimo, 2021-09-06:
On a Raspberry pi 4 using Ubuntu 20.04 64bit, step 5 comes up with this:

```
cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true
-- Checking internet connection...
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.49.0
-- Setting Unix configurations
-- Info: Building with CUDA requires CMake v3.8+
-- The CUDA compiler identification is unknown
CMake Error at CMake/cuda_config.cmake:3 (project):
  No CMAKE_CUDA_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CUDACXX" or the CMake cache entry CMAKE_CUDA_COMPILER to the full
  path to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMake/global_config.cmake:73 (include)
  CMakeLists.txt:44 (global_set_flags)

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/librealsense/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/librealsense/build/CMakeFiles/CMakeError.log".
```
As I'm not very used to cmake, yet looking into both .log files, I couldn't find any reference to an erroneous attempt to call a CUDA compiler, see  CMakeError.log

```
Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/ubuntu/catkin_ws/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_96a1d/fast && /usr/bin/make -f CMakeFiles/cmTC_96a1d.dir/build.make CMakeFiles/cmTC_96a1d.dir/build
make[1]: Entering directory '/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_96a1d.dir/src.c.o
/usr/bin/cc   -DCMAKE_HAVE_LIBC_PTHREAD   -o CMakeFiles/cmTC_96a1d.dir/src.c.o   -c /home/ubuntu/catkin_ws/build/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_96a1d
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_96a1d.dir/link.txt --verbose=1
/usr/bin/cc  -DCMAKE_HAVE_LIBC_PTHREAD    -rdynamic CMakeFiles/cmTC_96a1d.dir/src.c.o  -o cmTC_96a1d 
/usr/bin/ld: CMakeFiles/cmTC_96a1d.dir/src.c.o: in function `main':
src.c:(.text+0x48): undefined reference to `pthread_create'
/usr/bin/ld: src.c:(.text+0x50): undefined reference to `pthread_detach'
/usr/bin/ld: src.c:(.text+0x5c): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_96a1d.dir/build.make:87: cmTC_96a1d] Error 1
make[1]: Leaving directory '/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_96a1d/fast] Error 2

Source file was:
#include <pthread.h>

void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/ubuntu/catkin_ws/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_d43df/fast && /usr/bin/make -f CMakeFiles/cmTC_d43df.dir/build.make CMakeFiles/cmTC_d43df.dir/build
make[1]: Entering directory '/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_d43df.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_d43df.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.16/Modules/CheckFunctionExists.c
Linking C executable cmTC_d43df
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_d43df.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_d43df.dir/CheckFunctionExists.c.o  -o cmTC_d43df  -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_d43df.dir/build.make:87: cmTC_d43df] Error 1
make[1]: Leaving directory '/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_d43df/fast] Error 2
```

No idea where to prevent the CUDA related copmiler call.

GBT16, 2021-09-13:
> @MartyG-RealSense Thank you very much for the instructions above.
> 
> Unfortunately, I am still having trouble importing pyrealsense2. Here are the steps I followed:
> 
> 1. Download the zip file from https://github.com/IntelRealSense/librealsense/releases/. I am using version 2.38.1
> 2. Extract the file and create a dir called build and cd into it.
>    My directory structure for the build dir is `/home/rakshak/git_repo/librealsense-2.38.1/build `
> 3. Run the CMake command to test to see if the build will work. I used this command
>    `cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true `
> 4. Still in the build dir. Run `make -j4 `and then `sudo make install`
> 5. Add`export PYTHONPATH=$PYTHONPATH:/usr/local/lib` `to` the end of my .bashrc and source it.
> 
> Now when I try to import it into python3, I get the following error.
> 
> ```
> rakshak@RakshakNano:~$ python3
> Python 3.6.9 (default, Jul 17 2020, 12:50:27) 
> [GCC 8.4.0] on linux
> Type "help", "copyright", "credits" or "license" for more information.
> >>> import pyrealsense2 as rs
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
> ModuleNotFoundError: No module named 'pyrealsense2'
> ```
> 
> I can see the librealsense2* files in `/usr/local/lib/`
> 
> ```
> rakshak@RakshakNano:/usr/local/lib$ ls
> cmake                librealsense2-gl.so         librealsense2.so.2.38    python3.6
> libfw.a              librealsense2-gl.so.2.38    librealsense2.so.2.38.1
> libglfw3.a           librealsense2-gl.so.2.38.1  pkgconfig
> librealsense-file.a  librealsense2.so            python2.7
> ```
> 
> I have also tried rebooting the Nano, still no luck.
> 
> Let me know if you need any more information.

hi @MartyG-RealSense I'm trying to install on python3.7 but I'm facing error on step 3 like this can you help me please?

[image]

what should I do?

MartyG-RealSense, 2021-09-14:
@GBT16  When you created the build folder with **mkdir build && cd build**, can you confirm please whether you were in the librealsense folder that has a list of files and folders in it when you inputted that command?

[image]

GBT16, 2021-09-14:
@MartyG-RealSense I already fix the problem by using lower version since I'm using Ubuntu 18.04 LTS. Thanksfor your help

MartyG-RealSense, 2021-09-14:
That's great to hear that you found a solution, @GBT16 - thanks for the update!

Danilich1994, 2022-02-15:
Hello, @MartyG-RealSense . I came to this issue topic from [HERE](https://support.intelrealsense.com/hc/en-us/community/posts/1500000077181-Realsense-Install-Ubuntu-20-04-Raspberry-PI-4). I want to install librealsense with python wrapper on ubuntu server 20.04, so I was following written [guide](https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-706643114)  up there. On step 3 i got following errors - 
```
-- Could NOT find Vulkan (missing: VULKAN_LIBRARY VULKAN_INCLUDE_DIR)
-- Using X11 for window creation
CMake Error at /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:165 (message):
  Could NOT find X11 (missing: X11_X11_INCLUDE_PATH X11_X11_LIB)
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:458 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.18/Modules/FindX11.cmake:414 (find_package_handle_standard_args)
  third-party/glfw/CMakeLists.txt:220 (find_package)

-- Configuring incomplete, errors occurred!

```
Looks like librealsense needs graphical interface to be installed on linux server so Viewer can work, am I right? But I do not need viewer, I just want to obtain some images from D415 and send these through Ethernet to different pc using python script.
So,if my theory right, then how to install SDK with python wrapper to ubuntu server without graphical interface installed?

MartyG-RealSense, 2022-02-15:
Hi @Danilich1994  If you are building librealsense 'headless' in CMake without support for graphics then you can exclude support for graphics-based examples and tools from the build by including the build flags **-DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=false** in your CMake build instruction.  This will cause only textual examples such as **rs-hello-realsense** to be included in the SDK build and exclude graphical examples and tools such as the RealSense Viewer.

If you are not building on an Nvidia Jetson board or you are building on a PC that does not have an Nvidia graphics chip / video card then you should leave out the build term **-DBUILD_WITH_CUDA:bool=true** from the CMake build instruction, as CUDA support only works with computing devices / computers that have an Nvidia graphics GPU.

Danilich1994, 2022-02-15:
@MartyG-RealSense thank you for assistance. Shall I delete all files from build folder before trying to build again? I knew about **-DBUILD_WITH_CUDA:bool=true**, but I just putted **false** for this parameter, will it work? Or should I just erase this line?

MartyG-RealSense, 2022-02-15:
I would recommend deleting the contents of the build folder before the next build attempt, yes.

The **-DBUILD_WITH_CUDA** flag is false by default anyway, but defining it as false in the CMake instruction does not do any harm as far as I am aware.

Danilich1994, 2022-02-15:
@MartyG-RealSense  great, thank you!

Danilich1994, 2022-02-16:
> 

@MartyG-RealSense  Putting **-DBUILD_GRAPHICAL_EXAMPLES=false** did not help. The same errors occurred 
```
-- Could NOT find Vulkan (missing: VULKAN_LIBRARY VULKAN_INCLUDE_DIR)
-- Using X11 for window creation
```
That's what I'm passing into cmake:
```
cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3.9 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=false -DBUILD_WITH_CUDA:bool=false
```

MartyG-RealSense, 2022-02-16:
Thanks very much @Danilich1994 for the information that installing the OpenGL and X11 libraries helped in your particular case!

kithib, 2022-03-18:
Hi ,when I do cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=...
I get an error

-- Build files have been written to: /home/nvidia/librealsense/build/external-projects/pybind11
[ 12%] Performing download step (git clone) for 'pybind11'
Cloning into 'pybind11'...
fatal: unable to access 'https://github.com/pybind/pybind11.git/': gnutls_handshake() failed: The TLS connection was non-properly terminated.
Cloning into 'pybind11'...
fatal: unable to access 'https://github.com/pybind/pybind11.git/': gnutls_handshake() failed: The TLS connection was non-properly terminated.
Cloning into 'pybind11'...
fatal: unable to access 'https://github.com/pybind/pybind11.git/': gnutls_handshake() failed: The TLS connection was non-properly terminated.
-- Had to git clone more than once:
          3 times.
CMake Error at /home/nvidia/librealsense/build/external-projects/pybind11/tmp/pybind11-gitclone.cmake:66 (message):
  Failed to clone repository: 'https://github.com/pybind/pybind11.git'

CMakeFiles/pybind11.dir/build.make:89: recipe for target 'src/pybind11-stamp/pybind11-download' failed
make[2]: *** [src/pybind11-stamp/pybind11-download] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/pybind11.dir/all' failed
make[1]: *** [CMakeFiles/pybind11.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
CMake Error at wrappers/python/CMakeLists.txt:22 (add_subdirectory):
  add_subdirectory given source
  "/home/nvidia/librealsense/build/third-party/pybind11" which is not an
  existing directory.

CMake Error at wrappers/python/CMakeLists.txt:53 (pybind11_add_module):
  Unknown CMake command "pybind11_add_module".

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/librealsense/build/CMakeFiles/CMakeOutput.log".

And I install pybind11 by myself,but it alse occour.
how could i do with it?

yosis1969, 2022-08-05:
> 

I am running Ubuntu 18.04 on Jetson Nano 2GB and while building librealsense i encounter the following error:
`naz@naz-desktop:~/Downloads/librealsense-2.50.0$ mkdir build
naz@naz-desktop:~/Downloads/librealsense-2.50.0$ cd build
naz@naz-desktop:~/Downloads/librealsense-2.50.0/build$ cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=...
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking internet connection...
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.50.0
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Setting Unix configurations
-- Found usb: /usr/lib/aarch64-linux-gnu/libusb-1.0.so  
-- Building libcurl enabled
-- Found OpenSSL: /usr/lib/aarch64-linux-gnu/libcrypto.so (found version "1.1.1")  
-- using RS2_USE_LIBUVC_BACKEND
-- Configuring done
-- Generating done
-- Build files have been written to: /home/naz/Downloads/librealsense-2.50.0/build/external-projects/pybind11
Scanning dependencies of target pybind11
[ 12%] Creating directories for 'pybind11'
[ 25%] Performing download step (git clone) for 'pybind11'
Cloning into 'pybind11'...
Note: checking out '8de7772cc72daca8e947b79b83fea46214931604'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 8de7772c chore: prepare for the 2.6.2 release (#2821)
[ 37%] Performing update step for 'pybind11'
[ 50%] No patch step for 'pybind11'
[ 62%] No configure step for 'pybind11'
[ 75%] No build step for 'pybind11'
[ 87%] No install step for 'pybind11'
[100%] Completed 'pybind11'
[100%] Built target pybind11
-- pybind11 v2.6.2 
-- Found PythonInterp: /home/naz/Downloads/librealsense-2.50.0/build/...  
CMake Error at build/third-party/pybind11/tools/FindPythonLibsNew.cmake:133 (message):
  Python config failure:

Call Stack (most recent call first):
  build/third-party/pybind11/tools/pybind11Tools.cmake:45 (find_package)
  build/third-party/pybind11/tools/pybind11Common.cmake:201 (include)
  build/third-party/pybind11/CMakeLists.txt:188 (include)

-- Configuring incomplete, errors occurred!
See also "/home/naz/Downloads/librealsense-2.50.0/build/CMakeFiles/CMakeOutput.log".
naz@naz-desktop:~/Downloads/librealsense-2.50.0/build$ 
`
I updated my cmake version to 3.19 and installed pybind11 again. However it can't seem to build the library. I even tried building older version like 2.47, 2.39 but they give the same error.

MartyG-RealSense, 2022-09-12:
Hi everyone,

The RealSense user @35selim has kindly shared at https://github.com/IntelRealSense/librealsense/issues/10891 a bash script that they have created for installation of librealsense and pyrealsense2.