realsenseai/librealsense #6965: Camera L515 is not detected on Jetson Nano

maxgrossin, 2020-07-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {L515} | 
| Firmware Version                   | (Open RealSense Viewer --> v2.36.0 | 
| Operating System & Version |   Ubuntu 18 | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                      | 
| Platform                                 | Jetson nano |                     | 
| Language                            |  python            | 

### Issue Description
Hi,

I'm currently working on a project using Jetson Nano and Intel Realsense L515. I managed to install the python wrapper on the Jetson Board, but while I was trying to gather the data using pipeline it doesn't work saying that no device is connected.
I tried to use the realsense-viewer software too but the camera is not detected here also.

Could you provide me some help there ?
Thanks a lot,

fredy1221, 2020-07-29:
similar issue with me:
https://github.com/IntelRealSense/librealsense/issues/6964

dorodnic, 2020-07-29:
Please review #6656
If you are building from source, please make sure to use `cmake .. -DFORCE_LIBUVC=ON`

maxgrossin, 2020-07-29:
@fredy1221 it is not the same issue as yours as the camera is not recognised by the realsense viewer too.

@dorodnic thanks, I will try this afternoon using the link issue and your recommandation, I will let you know :D

fredy1221, 2020-07-29:
After installing pyrealsense2 wrapper, it's no longer detected in realsense viewer

> @fredy1221 it is not the same issue as yours as the camera is not recognised by the realsense viewer too.
> 
> @dorodnic thanks, I will try this afternoon using the link issue and your recommandation, I will let you know :D

fredy1221, 2020-07-29:
After installing pyrealsense2 wrapper, it's no longer detected in realsense viewer

maxgrossin, 2020-07-29:
Ok issue fixed thanks for your help @dorodnic !

@fredy1221 the way of I managed it is : 
- Uninstall the last cmake I made for Python on your build file : `sudo make uninstall`
- Recreate the build file and I use this cmake command : `cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DFORCE_RSUSB_BACKEND=ON -DCMAKE_BUILD_TYPE=Release -DFORCE_LIBUVC=ON -DPYTHON_EXECUTABLE=/usr/bin/python`
- Than `make -j4` and `sudo make install`

At this point my camera was still detected using the `rs-enumerate-devices` but in order to use the pyrealsense2 lib in my Python project I had to copy the pyrealsense2.so file from the build/wrappers/python folder using `sudo cp pyrealsense2.so /usr/local/bin`

Finally, do not forget to execute this command on your Python script folder : `export PYTHONPATH=$PYTHONPATH:/usr/local/lib ` 

I hope it will work for you :smile:

closed 2020-07-29 by maxgrossin

fredy1221, 2020-07-29:
Thanks, it actually worked when I added the `-DFORCE_LIBUVC=ON`